KLayout/klayout #161: Can't set keyboard shortcuts in 0.25.4 for mac

tlmassey, 2018-08-31:
I just installed klayout v0.25.4 on a new computer (Mac OS 10.13.6) and couldn't set my keyboard shortcuts.  The option to change anything was grayed out.  I downgraded to v0.23.11, close to what I was running on my old computer, and I can set the key bindings.  I can also successfully set key bindings in v0.24.10.  It seems this is an issue isolated to v0.25.

Thanks for any help or suggestions.

klayoutmatthias, 2018-08-31:
Hi,

thanks for reporting this issue. Do you have screenshot you could attach to this ticket?

Thanks,

Matthias

klayoutmatthias, 2018-08-31:
I tried to reproduce the issue with my VirtualBox image (10.13), but I cannot reproduce the behaviour. Please see the attached screenshot.

Matthias
[image]

tlmassey, 2018-08-31:
My mistake, I wasn't double clicking the item before trying to edit the
hotkey.  I was only single clicking.  It's working for me now.  Thanks.

klayoutmatthias, 2018-09-01:
Thanks ... I'll close the issue then.

closed 2018-09-01 by klayoutmatthias