KLayout/klayout #166: Internal error: ../../../src/db/db/dbPolygonTools.cc:489 _cut_polygon_internal (poly, line, right_of_line) was not true

lukasc-ubc, 2018-09-12:
In KLayout 0.25.4, MacOS:
* I created a layout in GDS
* ran a script to do an inversion
* Saved the GDS, and received the above error
* Saved the file as a DXF, and that works
* Closed and opened the DXF
* Tried saving again as a GDS, and same error

* I also tried to clean up the polygons by doing "Merge Shapes", and that didn't help.

[wg_loss_zep2.DXF.zip](https://github.com/klayoutmatthias/klayout/files/2376975/wg_loss_zep2.DXF.zip)

klayoutmatthias, 2018-09-16:
Hi Lukas,

This error means the polygon cut algorithm which reduces the number of points down to the maximum allowed number encountered a weired situation it cannot resolve. If you used boolean operations, these situations should not arise as the resulting polygons are merged. I'll check the test case. Thanks for providing it.

DXF works because it does not have a maximum vertex count. The same holds for OASIS and CIF.

Best regards,

Matthias

klayoutmatthias, 2018-09-16:
Hi Lukas,

I was able to fix the issue. The reason was actually a rounding issue: the small gaps that appear in the output are sometimes closed when the polygon cut algorithm splits the polygons due to cut-point snapping. This induces a topology change that the algorithm is not expecting. I can fix this by using a higher precision internally. 

I can now save your layout even with a max vertex count of 4 which gives a funny fractured layout. But XOR-wise it's (almost, within 1 DBU) the same than the original :-)

Matthias

commit e9de4252 (2018-09-26): Fixed #166 (internal error when writing GDS file)

This fix consists of computing the intersection
points in the split procedure with higher resolution
to avoid topology changes due to snapping on the cut
line.