KLayout/klayout #178: Maker Database Browser, short-cut keys don't work

lukasc-ubc, 2018-10-01:
After running a DRC, and when an error is selected, many of the menu options don't work, e.g., Descend, Show As New Top. Neither via hotkeys, nor via the menu.  Other things like Zoom Fit work.

The fix is to go to the Marker Database Browser, and unselect the error, by click in any blank region of the window (under Directory, or Markers). Once unselected, KLayout behaves as normal.

klayoutmatthias, 2018-10-01:
Hi Lukas,

this time I'm able to reproduce the issue :-)

There is an explanation and a solution. But first I think you have selected "Marker Database Top Cell" for the context (Marker Browser -> Configure). If not, I have to check again. If that is true, please continue reading ...

In "Marker Database Top Cell" mode, the marker browser forces the view to stay within the cell given by the marker database. That's why "Descend" does not work. Basically, when the DRC is run, the tool records the name of the top cell the DRC is run on. When the context is "Marker Database Top Cell" the browser will select this cell before displaying the marker. If you have loaded a layout with a different cell you will get a warning. This mode is supposed to ensure, the markers get displayed in their original reference frame. But that's not always what is intended. That's why there are other modes.

For example, you could chose "Any cell": then the marker browser will then accept any top cell cell as the reference cell for the marker database regardless of it's name and display the marker in this cell. In this mode you can safely descend into the hierarchy and the markers will still be displayed in the top cell's reference frame. "Current cell" will accept the current cell and display the markers within the frame of the current cell - this is bad when descending because the markers will be carried into the child cell and display incorrectly. 

So I'd recommend "Any cell". With this mode, "descend", "ascend" and the other hierarchy navigation features should work.

Best regards,

Matthias

klayoutmatthias, 2018-10-01:
I already closed the issue, because I try to keep the ticket list a little tidy - if this explanation isn't helpful, please feel free to reopen it. 

Thanks,

Matthias

closed 2018-10-01 by klayoutmatthias