KLayout/klayout #187: MacOS build for 0.25.5

Kazzz-S, 2018-10-21:
@klayoutmatthias

Thank you for making the minor release of 0.25.5!

I've built different binary packages for MacOS as usual (*) and they are available below.
https://www.dropbox.com/sh/3prpbm50retju0z/AAB73ejTWwU9wpt5xBe7_o5la?dl=0

Srl.No.|Target OS|Build Env. OS(1)|Qt version(2)|Binary Package
:-----:|:-------:|:--------------:|:-----------:|:-------------
1|Yosemite(10.10)|Yosemite(10.10)|5.10.1(3)|klayout-0.25.5-MacOSX-Yosemite-1-Qt5101mp.dmg
2|El Capitan(10.11) Sierra(10.12)|El Capitan(10.11)|5.11.2|klayout-0.25.5-MacOSX-ElCapitan-1-Qt5112mp.dmg
3|HighSierra(10.13) Mojave(10.14)|HighSierra(10.13)|5.11.2|klayout-0.25.5-macOS-HighSierra-1-Qt5112mp.dmg

Notes:
(*) with new simplified DMG installer
(1) using Ruby and Python bundled with OS
(2) using packages from MacPorts
(3) looks the last available version on Yosemite

Best regards,
Kazzz

klayoutmatthias, 2018-10-21:
Hi Kazunari,

thanks a lot! I have uploaded them already.

Best regards,

Matthias

thomaslima, 2018-10-22:
Dear Matthias,

I have the build for v0.25.5 MacOS HighSierra with python 3.7 from brew. https://www.dropbox.com/s/s31v0a9e5gdgb4e/qt5.pkg.macos-HighSierra-B37-release-0.25.5-2bcacdf.dmg?dl=0
https://www.dropbox.com/s/3rf0fw4mxr50mrz/qt5.pkg.macos-HighSierra-B37-release-0.25.5-2bcacdf.dmg.md5?dl=0

Best,
T

klayoutmatthias, 2018-10-22:
> brew and qt 5.11.2 (built via travis btw).

Hi Thomas,

thanks a lot ... I have received a build from kazz already. Does it  
make sense to put both on the server? If so, how do I best describe  
the differences?

And one more thing: I'm sorry to have hidden your 0.26.0.dev4 Python  
module - there is an actual issue I don't know how to overcome: if I  
deploy .dev4 for example, twine will tell me that there is .dev4  
already. So I had to increment the "dev" version. But this will create  
a new package without references to your ".dev4" deliverables.

Is there a way to lets say "add" file to a package through twine? Or  
is the only option a single deployment from travis?

Thanks,

Matthias

Kazzz-S, 2018-11-27:
Close this ticket since version 0.25.6 has been released.

closed 2018-11-27 by Kazzz-S