KLayout/klayout #201: MacOS build for 0.25.6

Kazzz-S, 2018-11-27:
Place holder

thomaslima, 2018-12-05:
I'd like to share my experimental version with python 3.7.1 from home-brew and Qt5.11.2. https://www.dropbox.com/s/nhhvzim7zyohae3/qt5.pkg.macos-HighSierra-B37-release-0.25.6-5ab1734.dmg?dl=0

Sorry about delay.

klayoutmatthias, 2018-12-09:
Hi Thomas,

no worries … do you like me to upload it to the server or do you think  
that is too experimental?

Best regards,

Matthias

thomaslima, 2018-12-10:
I think you could upload it. At least 5 colleagues of mine use it without problems.

Best,

Thomas Ferreira de Lima
tlima@princeton.edu<mailto:tlima@princeton.edu>

Zitat von Thomas Ferreira de Lima <notifications@github.com<mailto:notifications@github.com>>:

Kazzz-S, 2019-01-15:
Close this ticket since version 0.25.7 has been released.

closed 2019-01-15 by Kazzz-S