KLayout/klayout #202: python build failed on windows

jaspreetj, 2018-11-28:
I am trying to use klayout as an importable module in python.

Here is what I have tried so far:

1. using Anaconda (python 3.6.7) command prompt to **pip install klayout**

import klayout works but import klayout.db throws the following error
```python
Traceback (most recent call last):

  File "<ipython-input-3-d9cd5926da1b>", line 1, in <module>
    import klayout.db

  File "C:\Users\root\Anaconda3\lib\site-packages\klayout\db\__init__.py", line 1, in <module>
    import klayout.dbcore

ImportError: DLL load failed: The specified module could not be found.
```

2. Then I downloaded klayout source and tried to build it using command **python setup.py install** in anaconda prompt shell and the build failed
Here is the anaconda shell dump:

```cmd
running install
running bdist_egg
running egg_info
writing src/pymod/distutils_src\klayout.egg-info\PKG-INFO
writing dependency_links to src/pymod/distutils_src\klayout.egg-info\dependency_links.txt
writing top-level names to src/pymod/distutils_src\klayout.egg-info\top_level.txt
reading manifest file 'src/pymod/distutils_src\klayout.egg-info\SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching 'src\plugins\*\db_plugin\*.cc'
warning: no files found matching 'src\plugins\*\db_plugin\*.h'
writing manifest file 'src/pymod/distutils_src\klayout.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
running install_lib
running build_py
running build_ext
building 'klayout._tl' extension
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.13.26128\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -DHAVE_CURL=1 -DHAVE_EXPAT=1 -DMAKE_TL_LIBRARY=1 -IC:\Users\root\Anaconda3\include -IC:\Users\root\Anaconda3\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.13.26128\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.16299.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.16299.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.16299.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.16299.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.16299.0\cppwinrt" /EHsc /Tpsrc\tl\tl\tlVector.cc /Fobuild\temp.win-amd64-3.6\Release\src\tl\tl\tlVector.obj
tlVector.cc
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.13.26128\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -DHAVE_CURL=1 -DHAVE_EXPAT=1 -DMAKE_TL_LIBRARY=1 -IC:\Users\root\Anaconda3\include -IC:\Users\root\Anaconda3\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.13.26128\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.16299.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.16299.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.16299.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.16299.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.16299.0\cppwinrt" /EHsc /Tpsrc\tl\tl\tlScriptError.cc /Fobuild\temp.win-amd64-3.6\Release\src\tl\tl\tlScriptError.obj
tlScriptError.cc
c:\users\root\downloads\klayout-master\klayout-master\src\tl\tl\tlThreads.h(68): warning C4251: 'tl::Mutex::m_spinlock': class 'atomic::spinlock' needs to have dll-interface to be used by clients of class 'tl::Mutex'
c:\users\root\downloads\klayout-master\klayout-master\src\tl\tl\atomic/spinlock.h(31): note: see declaration of 'atomic::spinlock'
c:\users\root\downloads\klayout-master\klayout-master\src\tl\tl\tlException.h(132): warning C4251: 'tl::Exception::m_msg': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'tl::Exception'
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.13.26128\include\xstring(4273): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
c:\users\root\downloads\klayout-master\klayout-master\src\tl\tl\tlScriptError.h(60): warning C4251: 'tl::BacktraceElement::file': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of struct 'tl::BacktraceElement'
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.13.26128\include\xstring(4273): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
c:\users\root\downloads\klayout-master\klayout-master\src\tl\tl\tlScriptError.h(62): warning C4251: 'tl::BacktraceElement::more_info': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of struct 'tl::BacktraceElement'
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.13.26128\include\xstring(4273): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
c:\users\root\downloads\klayout-master\klayout-master\src\tl\tl\tlScriptError.h(137): warning C4251: 'tl::ScriptError::m_sourcefile': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'tl::ScriptError'
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.13.26128\include\xstring(4273): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
c:\users\root\downloads\klayout-master\klayout-master\src\tl\tl\tlScriptError.h(139): warning C4251: 'tl::ScriptError::m_cls': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'tl::ScriptError'
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.13.26128\include\xstring(4273): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
c:\users\root\downloads\klayout-master\klayout-master\src\tl\tl\tlScriptError.h(140): warning C4251: 'tl::ScriptError::m_context': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'tl::ScriptError'
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.13.26128\include\xstring(4273): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
c:\users\root\downloads\klayout-master\klayout-master\src\tl\tl\tlScriptError.h(141): warning C4251: 'tl::ScriptError::m_backtrace': class 'std::vector<tl::BacktraceElement,std::allocator<_Ty>>' needs to have dll-interface to be used by clients of class 'tl::ScriptError'
        with
        [
            _Ty=tl::BacktraceElement
        ]
c:\users\root\downloads\klayout-master\klayout-master\src\tl\tl\tlScriptError.h(72): note: see declaration of 'std::vector<tl::BacktraceElement,std::allocator<_Ty>>'
        with
        [
            _Ty=tl::BacktraceElement
        ]
c:\users\root\downloads\klayout-master\klayout-master\src\tl\tl\tlString.h(755): warning C4251: 'tl::Extractor::m_str': class 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>' needs to have dll-interface to be used by clients of class 'tl::Extractor'
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.13.26128\include\xstring(4273): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.13.26128\bin\HostX86\x64\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MD -DHAVE_CURL=1 -DHAVE_EXPAT=1 -DMAKE_TL_LIBRARY=1 -IC:\Users\root\Anaconda3\include -IC:\Users\root\Anaconda3\include "-IC:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.13.26128\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.6.1\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.16299.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.16299.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.16299.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.16299.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.16299.0\cppwinrt" /EHsc /Tpsrc\tl\tl\tlString.cc /Fobuild\temp.win-amd64-3.6\Release\src\tl\tl\tlString.obj
[... cut ...]
C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.13.26128\include\xstring(4273): note: see declaration of 'std::basic_string<char,std::char_traits<char>,std::allocator<char>>'
src\tl\tl\tlXMLParser.cc(35): fatal error C1083: Cannot open include file: 'expat.h': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Community\\VC\\Tools\\MSVC\\14.13.26128\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2
```

Any Ideas how can I get it working on a windows system?

klayoutmatthias, 2018-11-28:
Hi,

The whole thing is work in progress, so some additional work is required to get it running. Specifically some third party libraries need to be installed before:

To get the PyPI package running:

1.) Download the 3rd party package from https://www.klayout.org/downloads/klayout-bits-1.0.zip

2.) Unzip (e.g. under ```c:\klayout-bits-1.0```)

3.) Install klayout package

```pip install klayout```

4.) Add the 3rd-party packages to the path (sorry, lengthy):

```set
PATH=C:\klayout-bits-1.0\msvc2017\x64\curl\bin;c:\klayout-bits-1.0\msvc2017\x64\expat\bin;c:\klayout-bits-1.0\msvc2017\x64\ptw\bin;c:\klayout-bits-1.0\msvc2017\x64\zlib\bin;%PATH%```

When you want to build the package yourself, you need the same 3rd party package. For instructions how to build see here: https://github.com/klayoutmatthias/klayout#build-instructions-windows-msvc-2017

Matthias

jaspreetj, 2018-11-29:
It worked!! Thank you 👍

closed 2018-11-29 by jaspreetj