KLayout/klayout #203: Segfault on CentOS7 on exit

klayoutmatthias, 2018-11-28:
See discussion here: https://www.klayout.de/forum/discussion/comment/4820

Apparently the following happens:
* KLayout will destroy all top level widgets on exit
* But the QMultiInputContext plugin registers it's own QMenu (a top level widget)
* When KLayout exists and the plugin tries to delete it's QMenu, a double delete happens and the segfault occurs.

Solution: KLayout should only delete those top level widgets it actually owns. This is difficult however, as these Widgets may be created from Ruby which shuts down after the application object has been destroyed.

klayoutmatthias, 2018-11-28:
This is the code that creates the issue (layApplication.cc: 1476++):

```
  QWidgetList tl_widgets = topLevelWidgets ();
  for (QWidgetList::iterator w = tl_widgets.begin (); w != tl_widgets.end (); ++w) {
    delete *w;
  }
```

It is supposed to prevent the following issue:

1.) Open KLayout, run Macro Development IDE
2.) Enter: ```RBA::QPushButton::new("Push me").show```
3.) A button appears
4.) Close KLayout without hitting the button -> segfault

The proposed fix is to delete top level widgets only if created by Ruby.

klayoutmatthias, 2018-11-28:
Fixed with the above commit. Will be released with the next minor release.

closed 2018-11-28 by klayoutmatthias

catrlica, 2023-02-23:
I have install 0.28.5 and I get this error

commit 2b815ef7 (2018-11-28): Fixed #203 (segfault on exit on CentOS7)