KLayout/klayout #204: Python library linking on OpenBSD

bentley, 2018-12-07:
Building klayout (git master) on OpenBSD fails in pya.

    c++ -pthread -Wl,-rpath,/usr/X11R6/lib -Wl,-rpath,/usr/local/lib/qt5 -Wl,-rpath-link,/usr/local/lib/qt5 -shared  -o libklayout_pya.so.0.26 pya.o pyaConvert.o pyaHelpers.o pyaInspector.o pyaMarshal.o pyaObject.o pyaRefs.o pyaUtils.o pyaModule.o pyaSignalHandler.o pyaStatusChangedListener.o  -lz /usr/local/lib//libpython3.6m.so -L/usr/ports/pobj/klayout-0.25.6/klayout-0.25.6/build-release/pya/pya/../.. -lklayout_tl -lklayout_gsi -L/usr/local/lib/qt5 -lQt5Designer -L/usr/local/lib -L/usr/X11R6/lib -lQt5PrintSupport -lQt5MultimediaWidgets -lpulse-mainloop-glib -lpulse -lglib-2.0 -lintl -lQt5Svg -lQt5Widgets -lQt5Multimedia -lQt5Gui -lQt5XmlPatterns -lQt5Network -lQt5Xml -lQt5Sql -lQt5Core -lGL
    c++: error: no such file or directory: '/usr/local/lib//libpython3.6m.so'

The problem is that build.sh sets PYTHONLIBFILE to `/usr/local/lib//libpython3.6m.so`, a file that does not exist. On OpenBSD shared libraries all have a filename of `libfoo.so.X.Y` (where X and Y are the library major and minor number). The linker will automatically pick the right one at link time when passed `-lfoo`, and ld.so will pick an appropriate shared library when loading the program.

Ideally this would use pkg-config. On OpenBSD `pkg-config --libs python-3.6m` returns `-L/usr/local/lib -lpython3.6m`. As far as I can tell there’s no way to pass these flags to build.sh.

klayoutmatthias, 2018-12-11:
Hi bentley, 

The build script pulls the library name from sysconfig and distuils in this order (whichever comes first):

  * ```sysconfig.get_config_vars('LDLIBRARY')[0]```
  * ```sysconfig.get_config_vars('LIBRARY')[0]```
  * ```distutils.sysconfig.get_config_vars('LDLIBRARY')[0]```
  * ```distutils.sysconfig.get_config_vars('LIBRARY')[0]```

If Python on OpenBSD supplies the name in another place, let me know. I'll put it into the build script too.

Until then, you can specify a specific library path with the "-pylib" build script option.

Regards,

Matthias

klayoutmatthias, 2020-02-22:
As there are no updates I don't know whether action is required. I'll close it for now.

closed 2020-02-22 by klayoutmatthias