KLayout/klayout #217: Feature request: Python Windows wheel using continuous integration

thomaslima, 2018-12-31:
This is related to #202 and #124 .

Request 1: Since apparently there is a clean way to build layout's pymod using MSVC (https://github.com/klayoutmatthias/klayout#build-instructions-windows-msvc-2017), and one of the wheels is already uploaded to pypi (python 3.6, windows 64-bit https://pypi.org/project/klayout/0.26.0.dev8/#files), I suggest we use azure pipelines or appveyor to automatically build these wheels. It will be faster and more reliable than using our own virtual machines.

Requested python versions: 2.7, 3.5, 3.6, 3.7. Win 32- (optional) and 64-bit. 

Request 2: Have the external libraries be included in the wheel. I think it is possible, judging by bumpy and scikit packages, but it might be a pain to figure out how to do it.

Once that's done, we can release a python package klayout-0.26.0.beta that works on the main versions of mac, linux and windows.

ejprinz, 2019-01-01:
Hi Thomas, I downloaded the Windows wheel and installed it with Windows Python 3.6. It installs but when I try to load klayout with:
import klayout.db as kl
I get an error:
ImportError: DLL load failed: The specified module could not be found.
Do I need to set a PYTHONPATH variable?

Thx, Erwin

thomaslima, 2019-01-02:
@ejprinz no, but you need to download some DLLs and include their paths in your PATH variable. Matthias explains well here: https://github.com/klayoutmatthias/klayout/issues/202#issuecomment-442618888

Maybe the solution for that is to include the three necessary DLLs into the wheel. Not sure there will be licensing concerns.

ejprinz, 2019-01-02:
> here: #202 (comment)

Hi Thomas, thanks for the hint. I installed the additional pieces and it
works now. Very amazing.

Happy New Year to you & Matthias !!

Best regards, Erwin

PR #218 (2019-01-04): Azure/windows wheel builds Py 3.5,3.6,3.7 win32 and win64
Closes #217 

## Contributions:

-  Now `pip install klayout` works for python >= 3.5 on windows!

- Building pip wheels for windows (including libcurl, expat, zlib and pthread DLLs). I had to manually patch the wheel after the build to place the DLLs in the klayout module folder. *Python versions 3.5, 3.6, 3.7*, using MSVC++ 14 in *both Win 32 and 64* architectures. 3.4 and below would require MSVC++ 10.0, but I haven't tested them.

- I uploaded all wheels to pypi under version 0.26.0-dev10, excepting cp36 x64, which was already there [... cut]
closed 2019-01-04 by klayoutmatthias