KLayout/klayout #223: MacOS build for 0.25.7

Kazzz-S, 2019-01-15:
@klayoutmatthias

Thank you for making the maintenance release of 0.25.7.

I've built different binary packages for MacOS as usual (*) and they are available below.
https://www.dropbox.com/sh/kd9wlj9wdb3481a/AAD2LFYv2Otqrum1B4_2ZRvja?dl=0

<< Source ID = https://github.com/Kazzz-S/klayout/commit/8d572518ec5208b6b632bbcd8e56b126a98786be >>

Srl.No.|Target OS|Build Env. OS(1)|Qt version(2)|Binary Package
:-----:|:-------:|:--------------:|:-----------:|:-------------
1|Yosemite(10.10)|Yosemite(10.10)|5.10.1(3)|klayout-0.25.7-MacOSX-Yosemite-1-Qt5101mp.dmg
2|El Capitan(10.11)|El Capitan(10.11)|5.11.3|klayout-0.25.7-MacOSX-ElCapitan-1-Qt5113mp.dmg
3|HighSierra(10.13) Mojave(10.14)|HighSierra(10.13)|5.12.0|klayout-0.25.7-macOS-HighSierra-1-Qt5120mp.dmg

<< Two listed below are experimental >>

Srl.No.|Target OS|Build Env. OS(4)|Qt version(2)|Binary Package
:-----:|:-------:|:--------------:|:-----------:|:-------------
4|Sierra(10.12)|Sierra(10.12)|5.12.0|klayout-0.25.7-macOS-Sierra-1-Qt5120mp.dmg (5)
5|Mojave(10.14)|Mojave(10.14)|5.12.0|klayout-0.25.7-macOS-Mojave-1-Qt5120mp.dmg

Notes:
(*) with new simplified DMG installer
(1) using Ruby and Python bundled with OS
(2) using packages from MacPorts
(3) looks the last available version on Yosemite
(4) built in the same environment as the target OS
(5) **Ruby is disabled.** Please refer to https://github.com/Kazzz-S/klayout/issues/32 for the reasons.

Best regards,
Kazzz

lukasc-ubc, 2019-01-28:
I notice that on the new OSX, there is an option for Light or Dark mode.
[image]

KLayout seems to work with both, but it is difficult to work with Dark mode.  

There is a way to force Light mode, described here:
https://developer.apple.com/documentation/appkit/nsappearancecustomization/choosing_a_specific_appearance_for_your_app

Perhaps we make this change to the distribution, namely add to Info.plist:

    <key>NSRequiresAquaSystemAppearance</key>
    <true/>

lukasc-ubc, 2019-01-28:
Hello,

I also notice that this version is using Python 2.7, rather than 3.  (klayout-0.25.7-macOS-HighSierra-1-Qt5120mp.dmg)

My impression from the work of Thomas was that we were moving to have it run on version 3...

thanks

Lukas

thomaslima, 2019-01-28:
Thanks @lukasc-ubc for the suggestion. Can you share a screenshot of your layout in dark mode? 

Sorry I missed this version, haven't had the time yet.

Matthias hasn't uploaded the py3 version on the website since the .25.3 version. Its build steps are automated using Travis, but I need to figure out a way to upload them automatically to some place that is not my dropbox.

lukasc-ubc, 2019-02-21:
Hi @thomaslima 

Dark mode screenshot:

[image]

Regarding Python 3 and Mac OS version, can you post your build somewhere?

thanks
Lukas

Kazzz-S, 2019-02-26:
Thank you @lukasc-ubc for your suggestion.

> Perhaps we make this change to the distribution, namely add to Info.plist:
> 
> ```
> <key>NSRequiresAquaSystemAppearance</key>
> <true/>
> ```
This works in Mojave.
[image]

Kazzz

Kazzz-S, 2019-06-25:
Close this ticket since **0.25.8** and **0.25.9** were released.

Kazzz

closed 2019-06-25 by Kazzz-S