KLayout/klayout #231: Proposal: Standardize Python's version in KLayout

thomaslima, 2019-02-04:
Some SALT package developers use Python as their scripting language. Unfortunately, Python still has a lingering backwards-compatibility issue with version 2 vs. 3. By default, klayout's build script uses the system python's libraries to support its script interpreter. This results in three different versions of python for linux (<=3.5), mac (2.7) and Windows (3.5). As a result, SALT package managers have to support a wide variety of versions.

I propose to fix a version (preferably 3.6 or later) so that SALT packages can safely use python3's syntax and all the cool features of 3.6 (which has support until the end of 2021).

Maybe we can do the same for ruby, although I am not sure it is quite necessary.

What do you think?

thomaslima, 2019-02-04:
FYI @klayoutmatthias Travis just granted us a build-time extension, so the macOS builds should finish normally on travis.

I think Azure can deal with windows builds (32 and 64-bit), but I haven't worked on it. Linux seems a lot more difficult because there are many distros you support.

klayoutmatthias, 2019-02-06:
Hi Thomas,

the only reason there is Python 2 still around are the builds on CentOS6 and 7 for which (to my knowledge isn't a standard Python 3 package). On CentOS6 there also isn't a Qt >=4.8 which severly limits functionality. And I want to build against standard packages only - otherwise this isn't really a distribution.

Unfortunately I cannot skip this easily - CentOS6/7 (and RedHat 6/7) are still very common in industrial EDA flows. But I think we don't need to care too much about this limitation. If someone needs Python 3 support there is still the option for a suer to build it from source. On Linux that is fairly easy.

All other platforms support 3.6 unless I forgot something. So 3.6 minimum is fine with me.

Thanks,

Matthias

thomaslima, 2019-02-06:
To my knowledge Python 3 isn't a standard almost anywhere. But at least on Macs, which I know best, we're not bound by the system's python. My build downloads and embeds Python's framework and dynamic libraries inside the app. I thought Windows did the same. Is it not the case on Linux? What is the sys.path for klayout built on linux? 

Can we ship klayout with embedded python3 at least on windows and macs? Linux users could build from source.

klayoutmatthias, 2019-02-07:
Hi Thomas,

I think that's possible with Windows. 0.25.x isn't fully compatible with MSVC, so I need to build with MSYS. But they should offer Python 3. I'll try that on the next minor release. For Linux, Python 3 is available as option for most modern systems like Ubuntu. I'll try to pick this option whenever possible.

I think in general I should add an option in the Salt description file to specify the Python, Ruby, Qt ... requirements. This way, the package manager should be able to check whether a package can be used. I agree that just letting users fail isn't nice.

Thanks,

Matthias

thomaslima, 2019-02-09:
Thank you. That sounds like a good idea. For Python, it is enough to use x.y.* versions because they don’t change any syntax between minor releases.

But the problem with python3 is that it will probably break packages that people might have coded for python2.

Cheers
Thomas

lukasc-ubc, 2019-11-01:
Hi Thomas, Matthias,

It would certainly be wonderful if KLayout used Python 3 for Mac OSX.  In my field (photonics), I don't know anyone who is using a legacy Linux system with KLayout on it. Those that run legacy systems are probably running commercial EDA tools, and using KLayout to view their layouts. In such a case, they can transfer their files to another computer to view. 

Is there a way for you to find out if there are any users the old Linux builds?  Perhaps by distributing binary packages for Cent OS 8, and seeing the download stats?

Anyway, I have not been enjoying needing to make my code compatible with both Python 2 and 3.  Inevitably I am often introducing bugs where I break something on one or the other...

thanks!
Lukas

klayoutmatthias, 2019-11-17:
Should now be fixed by #387 (same topic) and is released in 0.26.1.

Can I close this issue?

thomaslima, 2019-11-27:
Ok for me.

closed 2023-05-31 by thomaslima