KLayout/klayout #235: "Cannot call non-const method on a const reference in Cell.shapes"

klayoutmatthias, 2019-02-24:
This ticket refers to the issue mentioned in the forum: https://www.klayout.de/forum/discussion/1191/cell-shapes-cannot-call-non-const-method-on-a-const-reference#latest

The code is:

```
import pya
lv = pya.LayoutView.current()
for obj in lv.object_selection:
  if obj.is_cell_inst(): 
    cell = obj.inst().cell
    for layer in lv.each_layer():
      layer_index = layer.layer_index()
      shapes = cell.shapes(layer_index)
```

Apparently it gives an error sometimes: "Cannot call non-const method on a const reference in Cell.shapes".

klayoutmatthias, 2020-02-23:
Can't reproduce on 0.26.x.

klayoutmatthias, 2020-10-10:
Reopened because another case is reported here: https://www.klayout.de/forum/discussion/1191/cell-shapes-cannot-call-non-const-method-on-a-const-reference

Still, I can't reproduce it. At least not with the simple script above.

klayoutmatthias, 2021-08-24:
I think this is fixed now as constness is handled differently in 0.27. Reopen if required.

closed 2021-08-24 by klayoutmatthias

lukasc-ubc, 2022-04-05:
@klayoutmatthias 

I just ran into this issue, in the following code:

```
    iter1 = topcell.begin_shapes_rec(layer_bb)
    while not(iter1.at_end()):
        subcell = iter1.cell()  
        subcell.name = subcell.name[:-2]            
        iter1.next()
```

which is fixed using your workaround:
```
        ly.cell(subcell.cell_index()).name = subcell.name[:-2]           
```

lukasc-dream, 2025-02-23:
@klayoutmatthias I encountered this error in a similar context as the original post.

https://github.com/SiEPIC/UBC-ELEC413-2025/blob/main/measurements/viewer.py#L271

klayoutmatthias, 2025-02-23:
Hi Lukas,

That may be a propagated issue - if the Cell is const, then "cell.layout()" will give you a const Layout and the will again give you a const Cell in 

```
cell.layout().cell(cell.cell_index())
```

So in the end, this redirection may be useless.

You can check if "cell._is_const_object()" gives True - in that case, the problem the nature of the incoming cell object is the issue (whatever reason for). This basically allows tracing back, where the problem comes from.

Or, in case you're tired of debugging, just use

```
cell_nc = cell._const_cast()
```

(available since version 0.29.6) the get rid of the constness flag.

Matthias

lukasc-ubc, 2025-02-24:
The redirection works every time so far, whereas without, it doesn't ever.  I did not see any inconsistent behaviour.

Trying your suggestion of tracking it back using _is_const_object(), I see True in the following:

    iter = layout.top_cell().begin_shapes_rec(layer_index)
    while not iter.at_end():
        if iter.shape().is_text():
            text = iter.shape().text.string
            if text == target_text:
                print(f"is iter cell const object? {iter.cell()._is_const_object()}")
                return iter.cell()
        iter.next()

if I append:
```
    print(f"is iter cell const object? {layout.cell(iter.cell().name)._is_const_object()}")
```
then I get True, False.

Which fixes it.

Is this because begin_shapes_rec returns a Const, hence iter becomes a Const, and then cell becomes a Const?  Whereas layout.cell() returns a non-Const.

However, if I put in both, in the opposite order:
```
    print(f"is iter cell const object? {layout.cell(iter.cell().name)._is_const_object()}")
    print(f"is iter cell const object? {iter.cell()._is_const_object()}")

```
then I get False, False.

This was unexpected.

My solution is to use 
```
   return layout.cell(iter.cell().name)
```

klayoutmatthias, 2025-02-24:
Hi Lukas,

well, to be honest, the whole constness affair is mainly there to reflect the C++ API, but the solution isn't perfect.

In short, your solution is good one. I'd just suggest (because names may not be unique) to use:

```
return layout.cell(iter.cell_index())
```

The basic idea is to prevent manipulation of cells during iteration. Hence, `iter.cell` returns a const object and you can fix the problem by either using "_const_cast()", or going through the cell index.

Problem is that Python actually does not have a notion of constness or non-constness. It only knows non-const objects. Inside KLayout, an object knows whether it needs to be const or non-const and refuses to execute non-const methods if const. But once you have bound a C++ object to a Python object (which happens in case of the Cell), it can only adopt one configuration and that will be const first, but switched to non-const if needed. This is why first asking for the non-const Cell binds the cell to Python in a non-const fashion and - as Cells remain bound - becomes non-const in the second call. Which admittedly is unexpected.

Constness can be useful and it can be a nuissance. It's a double-edged sword :(

Matthias