KLayout/klayout #238: dxf import takes ages

HoWilgh, 2019-03-15:
Hello,
I use klayout from time to time over years now to check mask layouts generated with python and ezdxf *).
With older versions (versions unknown, over a year ago) I could import the generated .dxf files with several 100MBs without any problem. With the current version it stops at some 'line' number and it takes unusual long for the import.
To give an example I generated a .dxf file (attached as many_lines.zip) which contains 100000 lines and was created with the attached .py file (attached as .py.txt). The .dxf file has a size of ~15MB and takes with klayout 0.25.8 and 0.25.4 ~1min. and 15 sec. In comparison Librecad and Layouteditor do import and visualize the file in <3sec. Importing larger .dxf files into klayout is therefore almost impossible (or takes ages).
BR,
HoWil
[many_lines.zip](https://github.com/KLayout/klayout/files/2970380/many_lines.zip)
[ezdxf_klayout_test.py.txt](https://github.com/KLayout/klayout/files/2970385/ezdxf_klayout_test.py.txt)
*)https://pypi.org/project/ezdxf/

Kazzz-S, 2019-03-15:
Hi HoWil,

With "Keep lines" option, it takes less than 1sec on Mac.
Does this do for you?

Regards,
Kazzz

[image]

HoWilgh, 2019-03-18:
> Hi HoWil,
> 
> With "Keep lines" option, it takes less than 1sec on Mac.
> Does this do for you?
> 
> Regards,
> Kazzz

Dear Kazzz,
Thank you very much!! It is working now as expected ... loading incredible fast again =).
I suggest to set this as default setting after installation.
Best regards,
HoWil

klayoutmatthias, 2019-03-20:
@kazzz - many thanks for solving this issue!

I don't know what's a good default. Everyone has different preferences. DXF is too broken to make one way the right one. Hence I just keep your latest settings. And the profile isn't reset by reinstallation. You should use the "Reset" button in the config page to do this.

I'll close this ticket.

Matthias

closed 2019-03-20 by klayoutmatthias