KLayout/klayout #241: [standalone Python module] TextGenerator.default_generator() returns None

Arold, 2019-03-19:
Hello,

I have tried the standalone Python module and it works really well. My only issue with it so far is that the `klayout.db.TextGenerator.default_generator()` returns `None`.

Best regard,

Arold
PS: Are you planing to release a Python 3.7 package on pypi?

klayoutmatthias, 2019-03-21:
Hi Arold,

thanks for this hint - I guess the default_generator isn't linked into the package.

Regarding the releases - the 0.26 version is still somewhat work in progress. I'll take a look into this later. PyPI deployment is not my favourite night job and I'm tired of digging through setuptools and disutils sources. Any volunteers for maintaining the PyPI releases? It's almost automated through Travis and Azure.

Matthias

closed 2019-03-22 by klayoutmatthias

thomaslima, 2019-03-22:
I’d like to help maintaining pypi but I’m currently on vacation. I did a lot of legwork of automating the wheel building. Last step is a Travis upload triggered at every tagged commit on master.

Closed #241<https://github.com/KLayout/klayout/issues/241>.