KLayout/klayout #243: Klayout not recognizing closed polygons from DXF

ashv10, 2019-03-19:
I am trying to convert an AutoCAD DXF file into GDS2 format using KLayout, for use as a photomask. On opening the file, several lines that were connected in AutoCAD appear disconnected in Klayout and the software does not automatically recognize the closed polygons. Is there a way to auto-detect closed polygons from a DXF file?

Kazzz-S, 2019-03-20:
Hello,

My suggestion is to set the "Merge accuracy" option shown below and attempt several times.
[image]

The DXF data below  **_looks_** a closed (at least, a designer believes so) quadrilateral in AutoCAD and alike, but there are gaps at three corners. I guess such a gap is very small in your case.  I have faced a similar case very often.
[image]

If I import this DXF into KLayout with the default "Merge accuracy" option, that is, zero, it will be as below.
[image]

Since the maximum gap is 0.2 in this case, by setting this value to "Marge accuracy," I can get a polygon as below. 
[image]

The DXF file is included in the ZIP file below.
[KLayout243.zip](https://github.com/KLayout/klayout/files/2988146/KLayout243.zip)

I hope this helps you.
Regards,
Kazzz

ashv10, 2019-03-20:
Thanks, but  that didn't fix the issue. I am still new to Klayout but if I understand correctly, shouldn't the software automatically hatch closed polygons to identify them? I have attached a file with a simple shape that I drew in AutoCAD. I opened it in Klayout using the 'merge accuracy' feature that you mentioned but I didn't see a hatched polygon like the one you've shown above.
[Klayout check.zip](https://github.com/KLayout/klayout/files/2989413/Klayout.check.zip)

Kazzz-S, 2019-03-20:
Hi,

My initial guess was wrong.
In the DXF file, you attached, the outer rectangle is drawn with a zero-width line.
So, KLayout imports it as a Path element by default. This is the reason.
[image]

By setting **LINE/POLYLINE** option below, 
[image]

You can get this.
[image]

Regards,
Kazzz

ashv10, 2019-03-20:
Thanks! That worked. 
Shouldn't the hatching be separate for the circle and the area outside it though? What I see is the same as the image you have included above - either the circle is hatched or the entire rectangle but the not the area in between the two.

klayoutmatthias, 2019-03-21:
@kazzz: many thanks again for helping!

@ashv10: in your DXF file I see a circle and a box. That are the graphics primitives and this is exactly what KLayout gives you. If you want only the area between the circle and the box to be filled, then you should draw every boundary as polyline and use the "Merge lines with width 0 into polygons" polyline mode. If you can use hatches they may be more reliable.

In general I need to say that while DXF is nice because it offers curved lines, it's definitely the *worst* format for making masks. Except for hatches and some primitives, DXF does not have the concept of "inside" and "outside" a polygon. But that is *absolutely crucial* for masks. IMHO DXF is a too big risk to take. If you search for "DXF guidelines for photomasks" you will see a lot of material about what issues to avoid. KLayout has many options to accommodate different drawing styles, but it will never be able the guess your intentions. So check the style you want to follow, configure KLayout accordingly, set up proper drawing guidelines and stick to it. Everything else will cost you.

Matthias

Kazzz-S, 2019-03-21:
@klayoutmatthias: This is a good opportunity for me to review various **Reader Options** provided. Your comments on DXF are educative! Let me quote them to convince my colleagues that _DXF is the worst format for making masks._

@ashv10: Here are some additional experiment results.
No.1: the same as above but circles are moved manually to show that three polygons are generated.
[image]

No.2: the same DXF file is imported with "Merge lines with width 0 into polygons" option.
[image]

No.3: importing an edited DXF file where the hatch has been removed. This must be the one you wanted.
[image]

Regards,
Kazzz

klayoutmatthias, 2019-03-22:
Hi kazz,

thanks for these tests ... actually the various options have evolved to accommodate different drawing styles. But there are too many of such styles and in some cases even the viewers get confused. 

Your material is too precious for keeping in hidden here. I'd like to cross-reference it from the forum. 

Thanks and best regards,

Matthias

ashv10, 2019-04-02:
@Kazzz-S @klayoutmatthias : Thank you for your inputs.
That fixed the problem!

klayoutmatthias, 2019-06-17:
Ok to close this issue?

ashv10, 2019-06-18:
Yes.

closed 2019-07-12 by klayoutmatthias