KLayout/klayout #247: Layout pane does not redraw

olcc, 2019-03-22:
Hello,
When I work on a layout and I run a script form the "Macro Development" window, the layout pane becomes uniformly gray, with nothing shown. I need to click on the layout pane for the layout to appear again. Sometimes I just need to hoover the mouse over the layout pane for it to appear again. But not always.
Should I run a redraw action at the end of my script? Which command? Is Klayout missing an automatic redraw action at the end of the scripts?
FWIW, the scripts in question are creating cells (with layout.create_cell()) that I will instantiate in the main layout (with the GUI).
Olivier (Klayout 0.25.8, Debian Linux)
PS: If needed I can try to take screenshots or try to build a minimal working example.

klayoutmatthias, 2019-03-28:
Hi olcc,

this is a know limitation - it happens because I have to disable redraw events during the execution of a script inside the debugger window. Redraw events can trigger script actions though crazy paths and this may finally end up in a deadlocked application.

It won't happen if you run a script from outside the macro IDE (e.g. bound to a menu). This is the preferred way to run scripts after you deployed them, so this problem should only occure during development of a script.

olcc, 2019-03-29:
I didn't know this, although I tried to search on the forum and in the issues before. I probably was not successful at searching. Thank you for the explanation.
Olivier

klayoutmatthias, 2019-04-02:
I'll close the issue now. It would be cool to have a debugger in a separate thread or process, but that's for later.

Matthias

closed 2019-04-02 by klayoutmatthias