KLayout/klayout #252: Basic library missing in standalone Python module

klayoutmatthias, 2019-04-02:
See https://www.klayout.de/forum/discussion/1210

Apparently basic.lib can't be seen from the standalone Python module.

klayoutmatthias, 2019-04-03:
The PR #255 will fix this issue but providing a new Python module which adds the libraries (currently: Basic).

Use it like this:

```python
import klayout.db
import klayout.lib

# says ['Basic']
print(klayout.db.Library.library_names())
```

The "lib" modules does not contribute symbols but will load the library code. It's not included in the db module to reflect to C++ code structure. 

**PLEASE NOTE:** The klayout.lib module has to be imported before the libraries are accessed the first time (i.e. before loading the first file or using the "db.Library" class). After the libraries have been established, loading the "klayout.lib" module does not have any effect. So it's recommended to do an "import klayout.lib" right at the beginning.

thomaslima, 2019-04-03:
@atait Check this out.
@lukasc-ubc It might interest you too.

thomaslima, 2019-04-03:
This basic library is a great example of how to define PCells in c++.

@klayoutmatthias Can you tell us how to add a python-defined library to the klayout.db.Library repository? I think there might be an example in SiEPIC-Tools https://github.com/lightwave-lab/SiEPIC_EBeam_PDK/blob/a929cf2b6c164b3b0ad90bb38b95cd5a30ed8830/klayout_dot_config/tech/EBeam/pymacros/SiEPIC_EBeam%20Library.lym#L1351-L1406

I feel that we should either (1) import basic.lib while importing .db or (2) make sure basic.lib is registered even if db.Library was accessed beforehand.

atait, 2019-04-03:
I ran into this problem just a few days ago while playing with sentinel/mask-design. It uses “TEXT” basic pcell. I just commented it out. A bad solution to this would be importing all installed libraries into .db. It would be very slow. Something like a .db command to load up some library by name. 

I have been working more on cell conversion on and off. I’m now able to run full sentinel in python, import it’s cells, convert them to phidl including ports, then use them in my phidl scripts. The reverse is easier because you can always run phidl anywhere. That’s already deployed in lygadgets. My application is that I don’t want to redesign N doped weights. If you think other people would have applications of this, let me know and I’ll put it in the to-release pile. 

-Alex

thomaslima, 2019-04-03:
@atait The difference is that Basic is a library implemented in c++ in klayout's source, so it is not "installed" anywhere. The python package should not look at anything installed in `~/.klayout`, in my opinion.

commit 86447506 (2019-04-04): Fixed #252 - updated setup.py and pipelines.
closed 2019-04-04 by klayoutmatthias