KLayout/klayout #261: Last `item.data()` of `pya.LayoutQuery` returns `None`

olcc, 2019-04-13:
This ticket refers to this issue mentioned in the forum: https://www.klayout.de/forum/discussion/1218/custom-query-in-a-python-script-bugs

In a query like below, the last `item.data()` unduly returns `None`. In particular, if there is only one cell instance matching the query, the result is `result  = [None]`, which is disappointing. 
```
query = "select cell.name from instances of .. where cell_name ~ 'CellName1'"
result = [item.data() for item in pya.LayoutQuery(query).each(layout)]
```

Olivier (KLayout 0.25.8, Debian Linux)

klayoutmatthias, 2019-04-16:
Hi Olivier,

thanks for this issue report. I can reproduce it, but that does not seem to be a systematic issue. With "search and replace" the result is fine. I'll check the Python integration code.

Regards,

Matthias

klayoutmatthias, 2019-04-16:
I guess I found the root cause. I'll give it some testing and then merge it.

commit 0bde4144 (2019-04-17): Fixed #261 (iterator must not be incremented while value is used)
commit e42b731e (2019-04-17): Merge pull request #262 from KLayout/issue-261

Fixed #261 (iterator must not be incremented while value is used)
commit d3c2c0fd (2019-06-18): Fixed #261 (iterator must not be incremented while value is used)
closed 2019-04-17 by klayoutmatthias