KLayout/klayout #269: Compiling on Redhat Linux

shaundurnan, 2019-05-20:


shaundurnan, 2019-05-20:
I am having the following error whilst compliling Klayout

system is "Red Hat Enterprise Linux Server release 6.4 (Santiago)"

Any assistance gratefully appreciated.

Shaun

/usr/bin/ld: gsiDeclQt_1.o: relocation R_X86_64_PC32 against undefined symbol `typeinfo for gsi::QFlagsAdaptor<Qt::KeyboardModifier>' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: ld returned 1 exit status
gmake[1]: *** [../../build-release/libklayout_gsiqt.so.0.25.8] Error 1
gmake[1]: Leaving directory `/home/sd09/Klayout/klayout-0.25.8/build-release/gsiqt'
gmake: *** [sub-gsiqt-make_default] Error 2

klayoutmatthias, 2019-05-20:
I guess something is wrong with your Qt installation and qmake doesn't specify the right options to gcc.

klayoutmatthias, 2019-06-16:
Does this issue still exist?

shaundurnan, 2019-06-17:
> CAUTION: This email originated from outside of the organization.

Hello Matthias,

                Unfortunately yes.

                I am using the tool on windows, but would prefer a compiled version on redhat.

                But I don’t have the experience to figure out what is wrong.

Thanks
Shaun

Shaun Durnan,
Principal Engineer,
Digital Power Lead,
Qualcomm Technologies International, Ltd.
Office: +44 1223 692280 (Int. 02280)
sdurnan@qti.qualcomm.com
[new logo]         [cid:image002.jpg@01D524F4.03F49B40]

klayoutmatthias, 2021-08-24:
Closing this issue as it's not a code base problem.

closed 2021-08-24 by klayoutmatthias