KLayout/klayout #270: Relative package definition

Grandement, 2019-05-20:
Hi all,
I am looking for a way to define the package index file relative to the grain.xml for local deployment. I would like to set the KLAYOUT_SALT_MINE environment variable to the package index file. The package index file should have a relative URL to the grain.xml of a package only. Paths with (.. /..) should be possible as well.
Moving of the complete package list could be handled easier.

klayoutmatthias, 2019-05-20:
You mean grain.xml URLs relative to package index file URL? The first sentence says package index relative to grain.xml which IMHO doesn't make sense.

Grandement, 2019-05-21:
Sorry you are right. Wrong order.

Grandement, 2019-05-21:
In this case the repository.xml would be just a folder.xml.
E.g.:
```
C:
`-- klayout
     |-- folder.xml
     |-- mypackage1
     |    `-- grain.xml
     `-- mypackage2
          `-- grain.xml
```

`KLAYOUT_SALT_MINE` <- C:\klayout\folder.xml

folder.xml:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<salt-mine>
 <salt-grain>
 	<name>mypackage1</name>
        <version>1.0.0</version>
 	<url>mypackage1</url>
 </salt-grain>
 <salt-grain>
 	<name>mypackage2</name>
        <version>1.0.0</version>
 	<url>mypackage2</url>
 </salt-grain>
</salt-mine>
```

When I copy the complete klayout folder to another location, it would be helpful when I can just change the `KLAYOUT_SALT_MINE` environment variable.

klayoutmatthias, 2019-06-17:
Hi,

I just tried to work on this issue, but I found it's already implemented this way. This is a repository.xml which works fine on my Linux installation:

```
<?xml version="1.0" encoding="utf-8"?>
<salt-mine>
  <salt-grain>
   <url>pkg1</url>
   <name>mypackage</name>
   <version>0.0</version>
   <title>Package 1</title>
   <license/>
   <icon/>
  </salt-grain>
  <salt-grain>
   <url>../pkg2</url>
   <name>mypackage2</name>
   <version>0.0</version>
   <title>Package 2</title>
   <license/>
   <icon/>
  </salt-grain>
</salt-mine>
```

I can put it somewhere, place two packages next (pkg1) and one level above (pkg2) and refer to the repostory.xml with "KLAYOUT_SALT_MINE=/path/to/repository.xml". pkg1 and pkg2 can be installed with these relative URLs.

Maybe your problem is already solved? Can I close this issue?

Matthias

Grandement, 2019-06-18:
Hi Matthias,

Thanks for checking it again. It works well in Linux. But I am not able to do it in Windows. I guess it has something to do with the path definition (\ ... /). What I am trying to do is to use a common source for linux and windows. E.g. the packages are located on a netdrive in a specific folder. My goal is that users from linux and windows can use the same index.xml file. Because of some technical reason I cannot use svn or git. It would be good when it is possible to use a ../ path definition for both systems.

Example:
```xml
 init
     `-- index.xml
packages
     |-- pkg1
     |    `-- grain.xml
     `-- pkg2
          `-- grain.xml
```

index.xml:
```xml
<?xml version="1.0" encoding="utf-8"?>
<salt-mine>
  <salt-grain>
   <url>../packages/pkg1</url>
   <name>mypackage1</name>
   <version>0.0</version>
   <title>Package 1</title>
   <license/>
   <icon/>
  </salt-grain>
  <salt-grain>
   <url>../packages/pkg2</url>
   <name>mypackage2</name>
   <version>0.0</version>
   <title>Package 2</title>
   <license/>
   <icon/>
  </salt-grain>
</salt-mine>
```

Grandement, 2019-07-18:
I did a workaround. I have setup symbolic links in linux which are working in Windows as well.
Right placement of the links give me a good solution.

I noticed that klayout used the root drive from the KLAYOUT_SALT_MINE variable.

For instance: KLAYOUT_SALT_MINE = Z:\fold1\fold2\index.xml

The Package Manager start at Z: in case of Windows. This not happens in Linux. Playing with the symbolic links as a jump from Z: resolve the path correctly.

closed 2019-07-18 by Grandement