KLayout/klayout #271: Feature Request:

demisjohn, 2019-05-31:
One of the most tedious things I find myself doing in KLayout is trying to *center* an instantiated cell at coordinates (0,0) or some well-known offset from there.  (ASML steppers always reference the centers of images/cells/patterns.)

Some KLayout functions that make this tedious:

- **Selection > Make Cell**:  Converts selected objects to a new cell, but with the Cell's internal origin & instantiation point set at the bottom-left of the cell (which is arbitrary based on the object extents in the selection - not really a useful design coordinate!).  I then have to move all objects inside the cell by half the cell's dimensions and then move the cell instantiation point.  Sometimes this is extremely complicated when the cell/objects are not symmetrically placed about the origin of the main layout!
- - If "Make cell" **maintained the origin of the original layout as the new instance's origin**, this would be extremely valuable and more intuitive (for me at least). It could pop up an options window to ask what the instance origin should be.
- - The drawback is that the origin might be outside the extent of objects - so does the bounding box for the cell extend all the way to include the origin, or just allow the + to lie outside the bounding box?  I think the latter.
- No visual marker of the current viewport's Origin.  Have to estimate by moving mouse around and looking at the coordinates.  Or I draw a dummy shape centered at (0,0).
- - If looking *into* a cell heirarchy, it should update the origin location marker, since a cell's origin is not the same as the top-level origin.
- Can't snap to origin.  Minor effect on the above problem.

Thanks!

lukasc-ubc, 2019-05-31:
I strongly agree with this suggestion. I’ve found these challenges myself, and I’ve seen many students struggle here too. A standard origin convention makes PDKs and teams easier to manage, so it would be nice if the tool facilitated this.

Lukas Chrostowski
Professor, ECE
University of British Columbia
604-822-8507
http://www.mina.ubc.ca/lukasc
http://siepic.ubc.ca

ejprinz, 2019-05-31:
This is highly nontrivial, because after centering the content in an
existing cell, all references to that cell in the layout have to be
adjusted to get the same top level outcome. If this cell has been rotated
or magnified, then the transformation in each reference has to be adjusted.
If you only need this for "make cell" then it is easier because no
references exist yet.

Best regards, Erwin

demisjohn, 2019-06-05:
(Not sure if you're saying that fixing this is non-trivial, or if the problem itself is non-trivial and is worth solving.)
@ejprinz : Aren't instantiations already handled any time one makes a change to a cell that has instantiations?  I am mostly looking at the `Make Cell` function, but presumably there are other functions that could automatically inherit the previous _origin_ as well.  I haven't run into these myself yet.  

> This is highly nontrivial, because after centering the content in an existing cell, all references to that cell in the layout have to be adjusted to get the same top level outcome. If this cell has [...]

@lukasc-ubc : Please post here if there are other functions that cause students to encountering this difficulty (w/r/to cell+global origins & instantiation points).  
Similar to you,  I'm running into this when I work with relatively new graduate students (also some companies) who submit jobs to us but don't actually run lithography themselves, so they don't know how important a cell/design's origin is to the manufacture of a reticle.

klayoutmatthias, 2019-06-05:
@demisjohn Have you looked into "Adjust Origin"? Select the cell in the cell tree and use "Edit/Cell/Adjust Origin" and select where you want to have the new center. This feature also allows adjusting the instances of the cell (on by default). The bounding box can be based on visible layers or on all layers.

klayoutmatthias, 2019-06-05:
One more thing: to see the origin, use File/Setup, Display/Background page and chose a style for the axes.

[image]

ejprinz, 2019-06-05:
@demisjohn:
I thought the issue being discussed is to center the contents of a cell with a Python script, rather than in the GUI. I am familiar with the KLayout GUI option, where you can select some content and center it. However, if you do this inside a cell, that cell changes and so if this cell is instantiated elsewhere it won't fit anymore the same way.
I wrote a Python script to center every cell in a layout, and my point is that it can be done, but one has to iterate through all instances of that cell and do the necessary transformations in each instance so things fit again after centering.

demisjohn, 2019-06-06:
@klayoutmatthias 
Thanks for pointing out these two functions.

The `Adjust Origin` command does not solve the Cell Instantiation problem, for two reasons:
1) it only has options for adjusting relative to the cell's Bounding Box - which again is not a useful design coordinate, as it is dependent on the arbitrary extents of objects inside the cell.  (I could put a dummy object inside the cell which is centered at (0,0) and symmetric (eg. a square/rectangle) as a workaround, similar effort as my current workarounds.)
2) I have to go _into_ the cell (make it the `Top Cell`), which means I can't see the original origin any more from before I used `Make Cell`.
(On a side note, I notice the Window that pops up upon `Adjust Origin` is actually called "_Align Cell_".)

Since `Adjust Origin` adjust the origin of the current viewport (effectively like doing a Move on all objects), as opposed to Adjusting the origin/instantiation point of a _Cell Instance_, I still think a new option for `Make Cell` would be the right place to address this.

The Axis options in Background does show the viewport's origin which is very helpful, thanks!  Now if only I could mouse-snap to the origin... but that is for another thread.

demisjohn, 2019-06-13:
Alternatively, this could also be a new function such as Selection > Adjust Origin (for selected cell instances)

klayoutmatthias, 2019-06-17:
Here is a proposal:

For adjusting the origin to the center:
[image]

For leaving the origin at the original position:
[image]

Matthias

demisjohn, 2019-06-17:
I like it.
Only suggestion is to make it explicit what will happen if the box is unchecked, eg. radio button instead.  But that is an aesthetic decision for you, if you did only what you propose here I would be very happy!

Like this:
[image]

demisjohn, 2019-06-17:
I see that you're reusing the existing `Move origin` dialog box, so the radio button is additional work that doesn't currently exist.  So this is fine as-is, thanks!

closed 2019-06-18 by klayoutmatthias

demisjohn, 2019-09-04:
Just tried this new version, it works very well.  Nice work, and thank you!

commit 46acf1ac (2019-06-18): Merge pull request #285 from KLayout/issue-271

Fixed #271 (proposal, more choices for the cell origin on 'make cell')
commit eecb6290 (2019-06-18): Fixed #271 (proposal, more choices for the cell origin on 'make cell')
commit af5c24d4 (2019-06-18): Fixed #271 (proposal, more choices for the cell origin on 'make cell')