KLayout/klayout #274: Provide uniform python interface between RecursiveShapeIterator and ObjectInstPath

olcc, 2019-06-15:
Hello,
`RecursiveShapeIterator` and `ObjectInstPath` are quite similar in some aspects, they both provide shape, layer, trans, dtrans, ... and for processing a list of shapes, they could be interchangeable. Code could be written to treat them indifferently. However this is not the case because their python interface are subtly different and thus incompatible:
- For a `RecursiveShapeIterator`, `element.shape()` and `element.layer()` are _methods_.
- For a `ObjectInstPath`, `element.shape` and `element.layer` are _attributes_.

Request: Offer a uniform python interface for both classes.

Olivier (Klayout 0.25.8)

klayoutmatthias, 2019-06-16:
Hi Olivier,

the problem is located a bit deeper. KLayout tries to provide a unified binding scheme for different languages and is rooted in Ruby. For Ruby, there is no difference between an attribute and a method - eventually, what looks like a read or write access are getter and setter methods. Ruby does not require the "()", so you don't see a difference between an attribute read access and a method call. 

Python has attributes and "properties" (with getters and setters, looking like attributes) and methods. KLayout tries to map the simple (and in my point of view elegant) Ruby concept to Python by providing properties when there is a non-parametrized getter method and corresponding setter method. Otherwise methods are created. Hence, "shape" is a *method* for ```RecursiveShapeIterator```, because there is no setter, but a property for ```ObjectInstPath``` because there is a setter and getter. 

To follow your proposal, I'd have a provide read-only properties too for methods that don't have parameters but a return value. That's feasible but I'm afraid of side effects as this change would take place deep inside the guts of the script language binding complex. I don't know for example, whether Python likes to have a method and a read-only property with the same name.

Matthias

olcc, 2019-06-16:
> I don't know for example, whether Python likes to have a method and a read-only property with the same name.

I would say it cannot, but this is above my level of knowledge of Python. Thank you for your explanation. So for the moment, I will make separate cases for RecursiveShapeIterator and ObjectInstPath.

closed 2019-07-12 by klayoutmatthias