KLayout/klayout #278: Selected shape object disappears (undesired garbage collection?)

olcc, 2019-06-16:
Hello,

In the following script, shape `s` disappears even though it is still bound to an identifier. Is it normal? Can it be improved?

```
import pya
import time

layout_view = pya.LayoutView.current() 
     
def get_selection():
  return layout_view.object_selection
  
# We suppose that the selection contains shapes
shapes = [opath.shape for opath in get_selection()]

s = shapes[0]
print("Shape: {}".format(s))  # Shape: box (x;y)
time.sleep(1)
print("Shape: {}".format(s))  # Shape: null
```
Note: A workaround is to keep a reference to `layout_view.object_selection`. But, as said, it is a workaround. The shape object should normally survive without this workaround, no?

klayoutmatthias, 2019-06-17:
Thanks for reporting this. It's kind of weird. Apparently I don't understand precisely what the list comprehension does with the reference count of Python objects. A simple ```print("anything")``` is sufficient to spoil the reference to the Shape objects.

The expanded form works well:

```
shapes = []
for opath in get_selection():
  shapes.append(opath.shape)
```

I wonder what's the difference between this form and ```[ opath.shape for opath in get_selection() ]```. That's going to be some awful debugging ...

Matthias

thomaslima, 2019-06-17:
Is there a way to print the reference count from C++?

klayoutmatthias, 2019-06-17:
OK, wasn't that difficult :)

klayoutmatthias, 2019-06-17:
@thomaslima It wasn't the reference count in fact (if you use Python from debug build it will track the counts). The issue was actually that the Shape object from ObjectInstPath#shape wasn't passed as copy but as pointer (the Python function generator by default treats ```const db::Shape &``` with copy semantics and ```db::Shape &``` as reference-to-member which bears the risk of lost references). In this case, this wasn't intended and is easy to fix by picking the right overload.

thomaslima, 2019-06-18:
I guess we can release a new version in pypi with this fix...

commit 2fd43ec6 (2019-06-18): Fixed #278 (lost reference to Shape object from ObjectInstPath)
commit fcec51e9 (2019-06-18): Merge pull request #284 from KLayout/issue-278

Fixed #278 (lost reference to Shape object from ObjectInstPath)
commit d0867d71 (2019-06-18): Fixed #278 (lost reference to Shape object from ObjectInstPath)
closed 2019-06-18 by klayoutmatthias