KLayout/klayout #287: Request: Return an ObjectInstPath from the RecursiveShapeIterator

olcc, 2019-06-18:
Hello,
The RecursiveShapeIterator provides many details, including the instantiation path of the shape. However there is still some work and difficulty for the user in order to build an ObjectInstPath for the shape from these parameters.

I would appreciate if there existed a method `RecursiveShapeIterator.opath()` that would return a suitable ObjectInstPath object for the shape.

This would make it possible to build a selection more easily, when the shapes come from a RecursiveShapeIterator.

Olivier

klayoutmatthias, 2019-07-12:
Hi Olivier,

I can't provide opath from ```RecursiveShapeIterator``` because that's crossing the db/GUI domain, but I can provide a ```ObjectInstPath``` constructor that takes an iterator and a cellview index.

As a quick workaround, you can define a corresponding constructor yourself. For example in Ruby:

```
module RBA

  class ObjectInstPath
    def ObjectInstPath.from_si(si, cv_index)
      oi = RBA::ObjectInstPath::new
      oi.top = si.top_cell.cell_index
      oi.cv_index = cv_index
      oi.clear_path
      si.path.each { |i| oi.append_path(i) }
      oi.shape = si.shape
      oi.layer = si.layer
      oi
    end
  end
  
end
```

Usage is like this:

```
lv = RBA::LayoutView::current
cv_index = lv.active_cellview_index

ly = lv.active_cellview.layout
top_cell = lv.active_cellview.cell

l2 = ly.layer(2, 0)

sel = []

lv.clear_object_selection

si = top_cell.begin_shapes_rec(l2)
while ! si.at_end?

  oi = RBA::ObjectInstPath::from_si(si, cv_index)
  lv.select_object(oi)

  si.next

end
```

I will provide a similar constructor natively then (it will be a variant of ```ObjectInstPath#new```.

Regards,

Matthias

closed 2019-07-13 by klayoutmatthias

olcc, 2019-07-13:
I think I understand what you explained.

> I will provide a similar constructor natively then

This will be convenient, thank you.
Olivier

commit 2d57a11f (2019-07-13): Fixed #287 (RecursiveShapeIterator to ObjectInstPath)

There is a new constructor for ObjectInstPath to
create one from a RecursiveShapeIterator.