KLayout/klayout #289: Segfault when calling LayoutView() in Python script

Juboal, 2019-06-21:
I am trying to save an image with `LayoutView.save_image()`
But merely creating a new LayoutView object results in a segmentation fault (`ERROR: Signal number: 11`).

I use Klayout 0.25.9  (installed with the `deb` file  from https://www.klayout.de/build.html) in a fully updated Ubuntu 18.04 LTS VM to call the python script in interpreter mode:
```
klayout -z -nc -r /path/to/test_layoutview.py
```

`test_layoutview.py`:
```python
import pya

#Test if calling LayoutView() works
lv = pya.LayoutView()  #This causes the seg fault
```

I have attached the full error message and the log of valgrind from running `valgrind klayout -z -nc  -r /path/to/test_layoutview.py .. >valgrind.log 2>&1
`

Any help would be highly appreciated :-)

[SegfaultErrorMessage.txt](https://github.com/KLayout/klayout/files/3314044/SegfaultErrorMessage.txt)
[valgrind.log](https://github.com/KLayout/klayout/files/3314362/valgrind.log)

klayoutmatthias, 2019-07-12:
This is copy of https://www.klayout.de/forum/discussion/comment/5665#Comment_5665
I'll keep the ticket to remind me not to segfault. For the solution see the link above.

klayoutmatthias, 2020-02-22:
There will be a more general solution in the next major release. This allows instantiating a LayoutView as a widget.

closed 2020-02-22 by klayoutmatthias