KLayout/klayout #292: MacOS build for 0.25.9

Kazzz-S, 2019-06-25:
@klayoutmatthias

Thank you for making the maintenance release of 0.25.9.

I've built different binary packages for **MacOS using OS-bundled Ruby & Python** as usual (*) and they are available below.
https://www.dropbox.com/sh/gx2hpt1lttz6get/AAD6nz9V-BBtfj7IQ0TWspl0a?dl=0

I hope different packages using **non-OS-bundled Ruby & Python (most likely Python 3.x)** are provided by different contributors.
Thank you for *YOUR* effort in advance.

<< Source ID = 20c910bb67209679b39220bfe0cb8478c3ed8b55 >>

Srl.No.|Target OS|Build Env. OS(1)|Qt version(2)|Binary Package
:-----:|:-------:|:--------------:|:-----------:|:-------------
1|Yosemite(10.10)|Yosemite(10.10)|5.10.1(3)|klayout-0.25.9-MacOSX-Yosemite-1-Qt5101mp.dmg
2|El Capitan(10.11)|El Capitan(10.11)|5.11.3(3)|klayout-0.25.9-MacOSX-ElCapitan-1-Qt5113mp.dmg
3|HighSierra(10.13) Mojave(10.14)|HighSierra(10.13)|5.12.4|klayout-0.25.9-macOS-HighSierra-1-Qt5124mp.dmg

<< Two listed below are experimental >>

Srl.No.|Target OS|Build Env. OS(4)|Qt version(2)|Binary Package
:-----:|:-------:|:--------------:|:-----------:|:-------------
4|Sierra(10.12)|Sierra(10.12)|5.12.4|klayout-0.25.9-macOS-Sierra-1-Qt5124mp.dmg (5)
5|Mojave(10.14)|Mojave(10.14)|5.12.4|klayout-0.25.9-macOS-Mojave-1-Qt5124mp.dmg (6)

Notes:
(*) with new simplified DMG installer
(1) using Ruby and Python bundled with OS
(2) using packages from MacPorts
(3) looks the last available version on Yosemite and El Capitan, respectively
(4) built in the same environment as the target OS
(5) **Ruby is disabled.** Please refer to https://github.com/Kazzz-S/klayout/issues/32 for the reasons.
(6) the same Ruby-related issue as (5) found with the latest security patch but manually rectified using 'Xcode-beta'.

Best regards,
Kazzz

Kazzz-S, 2019-06-25:
@klayoutmatthias 

I have skipped **0.25.8** (no specific reason; just busy), but they are available below.
https://www.dropbox.com/sh/qh0ycgu09avrtg9/AADbs9vqmzBKn-_ti4a0neama?dl=0 
Just for your record.

Best regards,
Kazzz

Kazzz-S, 2019-07-01:
@klayoutmatthias

I have rebuilt the package for **Sierra (10.12)** in which **Ruby is enabled**.
Ref. https://github.com/Kazzz-S/klayout/issues/32#issuecomment-507289546

Please replace the current No1-build with the No2-build below.
https://www.dropbox.com/sh/gx2hpt1lttz6get/AAD6nz9V-BBtfj7IQ0TWspl0a?dl=0

Srl.No.|Target OS|Build Env. OS(4)|Qt version(2)|Binary Package
:-----:|:-------:|:--------------:|:-----------:|:-------------
4|Sierra(10.12)|Sierra(10.12)|5.12.4|klayout-0.25.9-macOS-Sierra-**2**-Qt5124mp.dmg

[image]

Best regards,
Kazzz

klayoutmatthias, 2019-07-12:
Thanks at lot! I'll close this ticket as this is done.

closed 2019-07-12 by klayoutmatthias