KLayout/klayout #297: KLayout for MacOS Catalina

VictorShishkin, 2019-07-04:
Hello! A week ago I upgraded my OS to MacOS Catalina Beta, and now KLayout stopped working... I didn't expect there will be compatibility issue.
I really hope there will be update of KLayout for Catalina. It's a really great software!

[image]

calebjordan, 2019-07-11:
@Kazzz-S Do you have any ideas here? 

Some of us live on the bleeding edge and often suffer for it…

VictorShishkin, 2019-07-11:
These are problem details:

Process:               klayout [1153]
Path:                  /Applications/klayout.app/Contents/MacOS/klayout
Identifier:            de.klayout
Version:               0.25.9 (0.25.9)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           klayout [1153]
User ID:               502

Date/Time:             2019-07-12 02:31:23.136 +0900
OS Version:            Mac OS X 10.15 (19A501i)
Report Version:        12
Anonymous UUID:        6E0E9164-C6FE-4DFE-637F-FDEC9B7FDA56

Sleep/Wake UUID:       37E92EFE-68DB-451F-84C8-92C2B82A284F

Time Awake Since Boot: 16000 seconds
Time Since Wake:       15000 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  Library not loaded: /System/Library/Frameworks/Ruby.framework/Versions/2.3/usr/lib/libruby.2.3.0.dylib
  Referenced from: /Applications/klayout.app/Contents/MacOS/klayout
  Reason: image not found

Binary Images:
       0x10ce6a000 -        0x10ce75ffb +de.klayout (0.25.9 - 0.25.9) <FC66FD5D-BC69-3A45-AB07-4DDA2193E838> /Applications/klayout.app/Contents/MacOS/klayout
       0x11afcc000 -        0x11b05a81f  dyld (729) <B762DAD0-A0A4-3B06-A17D-BC9C1774A30B> /usr/lib/dyld
    0x7fff3ac01000 -     0x7fff3acf4fff  org.python.python (2.7.16 - 2.7.16) <2A013D34-6B3E-3526-80BB-6BC07D6F8868> /System/Library/Frameworks/Python.framework/Versions/2.7/Python

Model: MacBookPro12,1, BootROM 186.0.0.0.0, 2 processors, Intel Core i5, 2.9 GHz, 8 GB, SMC 2.28f7
Graphics: kHW_IntelIris6100Item, Intel Iris Graphics 6100, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1867 MHz, 0x80AD, 0x483943434E4E4E424C54424C41522D4E5544
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1867 MHz, 0x80AD, 0x483943434E4E4E424C54424C41522D4E5544
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x133), Broadcom BCM43xx 1.0 (7.77.97.1 AirPortDriverBrcmNIC-1424)
Bluetooth: Version 7.0.0d97, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0512G, 500.28 GB
USB Device: USB 3.0 Bus
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1

And here is a line from Apple system log:
Jul 12, 2019 2:31:23 Problem Reporter[1155]: assertion failed: 19A501i: libxpc.dylib + 87248 [5321C905-18A7-3566-9831-441CCFC2E794]: 0x89

I hope it's helpful

Kazzz-S, 2019-07-11:
@calebjordan Thank you for your comment. @VictorShishkin Thank you for providing detailed information.
I've tried to install **Catalina Beta** version in my VMware environment but failed in my first trial, unfortunately :-(  My real work environment is STILL Yosemite; far behind a bleeding-edge!

By the way, which binary package have you tried? I guess it's **klayout-0.25.9-macOS-Mojave-1-Qt5124mp.dmg**. Including this, those DMGs provided via the KLayout site are built to dynamically link OS-bundled Ruby and Python.

> Dyld Error Message:
> Library not loaded: /System/Library/Frameworks/Ruby.framework/Versions/2.3/usr/lib/libruby.2.3.0.dylib
> Referenced from: /Applications/klayout.app/Contents/MacOS/klayout
> Reason: image not found

Above error message suggests that the expected Ruby library (used in Mojave) is missing or most likely a different version of Ruby is bundled in Catalina. Could you please provide the version information about **Ruby and Python in Catalina** (I hope they exist) under:

``
/System/Library/Frameworks/Ruby.framework
``
and 
``
/System/Library/Frameworks/Python.framework
``

With warm regards,
Kazzz

calebjordan, 2019-07-12:
@Kazzz-S , does this help?

```
ls /System/Library/Frameworks/Ruby.framework/Versions 
2.6     Current
ls /System/Library/Frameworks/Python.framework/Versions
2.3     2.5     2.6     2.7     Current
```

Kazzz-S, 2019-07-12:
Hi @calebjordan,

Thanks. It looks that Ruby's version is **2.6**. 
Are you using KLayout with Ruby automation (RBA)?
I have built a special DMG on Mojave **disabling Ruby**, which is available at:
  https://www.dropbox.com/sh/qeny6se8fepgois/AACnthqtYTGxN78XhF02rav3a?dl=0
for experimental purpose. Please just try this package and see what would happen on Catalina.
If you do not need RBA, this can be a lazy solution :-)

With warm regards,
Kazzz
(Please refer to the next comment, too.)

Kazzz-S, 2019-07-12:
### About two different DMG packages ###
[image]
[image]

----
### Sample Images ###
[image]
[image]

-----
### Ruby cannot be used ###
[image]

----
### But Python (2.7.x) can be used ###
[image]
[image]
[image]

calebjordan, 2019-07-13:
@Kazzz-S, it worked! I only ever use the Python interpreter anyway. 

Thanks so much for the help!

VictorShishkin, 2019-07-17:
Awesome! It worked for me too! Thank you so much, @Kazzz-S!
Feels so good to work with KLayout again!

klayoutmatthias, 2019-07-22:
Good to hear this!  :-)

I'll close this issue then.

closed 2019-07-22 by klayoutmatthias