KLayout/klayout #305: CIF box positions are incorrect when a rotation is specified?

jhunacek, 2019-07-15:
I am having some trouble importing what I believe is valid CIF data into KLayout 0.25.9 (Ubuntu 18.04).  I have written a simplified test case below that demonstrates the behavior.  It seems boxes drawn with a rotation vector specified have their offsets scaled in a strange manner.  I was expecting rotated boxes to be centered on the same point as non-rotated boxes when the same position is specified (I haven't found anything in the spec that disagrees with this).  When importing into L-Edit (without the comments which don't seem to be supported there), this CIF data behaves as I expected.

```
DS 1 10 1;
9 test;
L TEST;

(Origin marker)
B 1 1 0,0;

(Draw two offset boxes without rotation)
B 10 1 20,-10;
B 10 1 20,10;

(Try drawing the same boxes with rotation)
(They appear nearly centered on the origin in Klayout, the offset is wrong?)
B 10 1 20,-10 5,-2;
B 10 1 20,10 5,2;

(If I scale the position by 10x the scale, then it appears where I expected it)
B 5 1 2000,-1000 5,-2;
B 5 1 2000,1000 5,2;

DF;
E
```

klayoutmatthias, 2019-07-21:
Thanks a lot for reporting this and thanks for the nice testcase!

Actually this is a bug. A workaround is to set the database unit in the reader options to 0.1 (in this case as the resoluton is 1/10 micron). But this is not an acceptable permanent solution.

The fix is easy. However, this bug is there for a long time already. I wonder how often rotated boxes are actually used. In fact, I have not seen CIF recently at all. 

Thanks,

Matthias

jhunacek, 2019-07-21:
Cool, thanks for the confirmation, I was worried I might be crazy!  Thank you for the workaround suggestion.  That does seem to improve the position, although the rotated boxes end up as parallelograms now which I don't think is expected. 

I can't speak to CIF usage in general, but I guess I am not surprised that it isn't common.  We have some legacy drawing code that generates it that we occasionally need to run (hence this issue), but our newer projects all output GDS files instead.

(Thank you so much for your work on KLayout, it really is a life saver!)

klayoutmatthias, 2019-07-24:
Thanks for your feedback :-)

BTW: if you want to try it, you'll find recent builds here: http://www.klayout.org/downloads/master. Disclaimer: the Windows .exe's aren't signed yet (my certificate is not available for these automated builds) and OpenSuSE 42.x isn't up to date (no Docker image available anymore).

Matthias

jhunacek, 2019-07-24:
Awesome, that works perfectly!  Thank you so much!

closed 2019-07-24 by jhunacek