KLayout/klayout #306: Round corners

martin-gustafsson, 2019-07-19:
Using "Edit/Selection/Round corners" produces unexpected results on shapes which bend in more than one direction.

Here are two paths before rounding:
[image]

Here they are again after rounding, both with the same parameters (72um outer radius, 50um inner radius, 100 points).
[image]

klayoutmatthias, 2019-07-21:
Hi,

I'm sorry, but I can't reproduce the issue. I have prepared some layout which looks roughly like the one you pasted:

[issue_306.gds.zip](https://github.com/KLayout/klayout/files/3415049/issue_306.gds.zip)

But all I get are reasonable images. Here for example outer radius = 225, inner radius = 200:

[image]

Could you provide a layout file and give some details about KLayout version and OS?

Matthias

martin-gustafsson, 2019-07-22:
Hi Matthias,

Thanks for looking into this. I wasn't able to reproduce the error in the file that you sent either.

In the attached GDS file though, layer 20 contains a path as drawn, and layer 30 contains a copy of it which I've rounded with 72um outer radius, 50um inner radius, and 100 points.

I'm using version 0.25.3 of KLayout on macOS High Sierra 10.13.6

Best wishes
Martin

On 7/21/19 5:12 PM, Matthias Köfferlein wrote:

Hi,

I'm sorry, but I can't reproduce the issue. I have prepared some layout which looks roughly like the one you pasted:

issue_306.gds.zip<https://github.com/KLayout/klayout/files/3415049/issue_306.gds.zip>

But all I get are reasonable images. Here for example outer radius = 225, inner radius = 200:

[image]<https://user-images.githubusercontent.com/25720841/61597140-ce2ab080-ac0c-11e9-8d4c-a100eaeeda22.png>

Could you provide a layout file and give some details about KLayout version and OS?

Matthias

klayoutmatthias, 2019-07-22:
Hi Martin,

I'm sorry, but I don't see an attachment :-(

Maybe this is because you replied by mail to this conversation thread. I think that GitHub strips attachments from mail replies. Maybe you can navigate to GitHub and drag and drop the file there. You will probably need to provide a .zip archive to make GitHub accept it. The issue link is: https://github.com/KLayout/klayout/issues/306

Thanks,

Matthias

martin-gustafsson, 2019-07-22:
Hi Matthias,

Sure! Here is the layout as a zipped GDS file.

Best wishes
Martin

[KLayout_rounding_demo.zip](https://github.com/KLayout/klayout/files/3418776/KLayout_rounding_demo.zip)

klayoutmatthias, 2019-08-17:
Thanks for the file ... I found some time to look into this issue.

Here is the analysis: The rounding function tries to guess whether the original shape already had some rounding and tries to undo this before applying the new rounding. In this case, because of the bent nature of the wire, the algorithm is fooled into thinking some rounding had been applied already. There is a numerical issue which eventually leads to this strange result.

I think the solution should be twofold: first, the algorithm needs to be a little more robust numerically and second, there needs to be an option to start with the actual polygon rather than the "unrounded" version.

Thanks for this test case,

Matthias

closed 2019-08-21 by klayoutmatthias