KLayout/klayout #320: DXF export creates AutoCAD incompatible files

dvase, 2019-08-08:
Using Klayout 0.25.9 I've discovered that DXF export creates a file that is 
unable to be read by AutoCAD when two cells names only differ by their capitalization, eg. 'lib' vs 'LIB'.

In Klayout the cells are recognized as two distinct entities and the cell name
capitalization survives the translation to DXF formatted text file, but AutoCAD 
apparently doesn't consider capitalization and is unable to read a file which has two blocks
of the same name.  As a side note, the Python library ezdxf also has a problem
reading these files, which is how I discovered the source of AutoCAD's inability
to read the DXF file.  While AutoCAD did not produce any meaningful output on 
why the file was unable to be read, ezdxf told me that there were multiple block 
definitions for certain cells.

I'm not quite sure what the fix should be.  Perhaps Klayout should produce a 
warning or error when it encounters these types of cell naming problems during 
DXF export?

Here is a trivial, simple example of a GDS file with its problematic DXF file output.
[cell_name_example.zip](https://github.com/KLayout/klayout/files/3482954/cell_name_example.zip)

Kazzz-S, 2019-08-10:
Hi,

I have asked my colleague to operate AutoCAD (on Windows-10) as below.
1. Create **AAA.dxf** as the first block and save it in a directory.
2. Create **aaa.dxf** as the second block and save it in a different directory.
3. Create a new design and insert **AAA.dxf** block. ==> This is OK.
4. Then insert **aaa.dxf** block. ==> This fails because the block **AAA** already exists in the design.

Below is a different interpretation by a non-AutoCAD tool, which is apparently incorrect.
[image]

Looking at these, I think an error should be flagged on as you suggested.
What do you think of the solution below?
[image]

Please refer to https://github.com/Kazzz-S/klayout/issues/35, too.

Regards,
Kazzz

klayoutmatthias, 2019-08-17:
To be frank this is not the only compatibility issue between GDS and DXF (in both directions). I guess there is much more to say about allowed cell names (e.g. allowed characters, encoding, maximum length, ...). 

Frankly, I'm a bit tired of reverse engineering a badly documented format like DXF. Specifically with this format, it's inevitable to establish a convention - not every GDS can be translated into DXF and vice versa. Some rules have to be observed when translation is supposed to go smoothly and caseless unique naming is just one point of this checklist. In most cases I've seen you don't even get an error message. Just wrong output. That's by far more dangerous in my perception.

Matthias

klayoutmatthias, 2020-04-09:
Here is the result of some experiments of mine.

Cell names:

* maximum length of 30 characters
* no special characters. Verified to work: "*", "_" and alphanumerics. Not working: "$", "?".

demisjohn, 2023-05-18:
Even within KLayout only (never going to Autocad), Klayout does some unusual things when saving as DXF, closing and re-opening the file:
- It adds a "0 0/0" layer
- upon reopening added another layer "L0D0_0"
- For one file from customers, it opened the generate file with cells "TOP" and "*viewport" or somesuch, but then I couldn't save changes due to "more than one TOP cell" - but obviously someone had sent a file with more than one topcell.

I do agree with Mattias - DXF is just a badly specified format, I have our customers avoid it as much as possible as it's impossible to fully support!