KLayout/klayout #333: Ruby/Python automation of "auto measure"

Kazzz-S, 2019-08-26:
Hi Matthias,

"Auto measure" is a handy feature of KLayout!
I wonder if this can be "doubly automated" by Ruby/Python.

Given: A layout design and a list of distinct points, each of which represents a single mouse click point in "auto measure."
Output: A set of annotations (rulers) inserted into the layout design.

Please refer to the attachment that contains a sample OASIS data, **AutoMeasure-01-Points.oas** and two image files.

In **Fig-01.png**, yellow rectangles (their center points) correspond to the mouse click points to draw "auto measures."
[image]

The result should be as shown in **Fig-02.png**.
[image]

In the sample data, there are such 1871 "sampling points."
To gain speed and to enable a precise click on each sampling point, automation is essential.

Those generated "cut vectors" are further used to control a machine to measure the width and the cross-sectional profile of the fabricated trace.

Is this functionality achievable with existing functions?
If not, I'll appreciate if you take this as a future enhancement request.

With best regards,
Kazzz

[AutoMeasure.zip](https://github.com/KLayout/klayout/files/3540353/AutoMeasure.zip)

----

[Initial discussion in KLayout Forum](https://www.klayout.de/forum/discussion/1310/ruby-python-automation-of-auto-measure)

klayoutmatthias, 2019-08-26:
Please see PR #335 for details about the implementation

Kazzz-S, 2019-08-27:
The suggested interface of the new feature in #335 is exactly what I want!

Thank you so much for your prompt action.

With best regards,
Kazzz

closed 2019-08-30 by klayoutmatthias

Kazzz-S, 2019-08-31:
Here are some updates :-)
I've implemented this new feature **LayoutView::create_measure_ruler()** in PYA.  
### It's fantastic!!!
It took **about 15 sec (on 64-bit Linux) to automatically create 1871 rulers**.
[image]

[AutoMeasure-Results.zip](https://github.com/KLayout/klayout/files/3562456/AutoMeasure-Results.zip)

Once again, thank you very much for providing the new feature.

With best regards,
Kazzz