KLayout/klayout #342: Feature request: Ruler Property Editor for "LayoutView::create_measure_ruler()"

Kazzz-S, 2019-09-01:
This is related to both:

1. https://github.com/KLayout/klayout/issues/333
and
2. https://github.com/KLayout/klayout/pull/335

The newly added feature **LayoutView::create_measure_ruler()** is extremely powerful and useful as I expected! Thank you very much for providing this.

To further enhance this feature, I would like to have an enhanced **Ruler Property Editor** with a GUI something like below. I hope the GUI is self-explanatory.

[image]

I'm preparing a separate test report regarding **LayoutView::create_measure_ruler()** https://github.com/KLayout/klayout/issues/341 where some issues captured through a kind of stress test will be described.

To me, it seems that the issues are (perhaps) related to the deep part of the "auto measure algorithm."
The implementation of **LayoutView::create_measure_ruler()** that wraps the algorithm looks fine.

I think this enhanced GUI should help debug the issues, too.

With warm regards,
Kazzz

klayoutmatthias, 2019-09-01:
Hi Kazzz,

thanks for this suggestion. I striving for generic functionality with as many use cases as possible. So I'd like to modify the proposal somewhat. Here is an idea:

[image]

"Swap points" would swap P1 and P2 (a function I wanted myself). "Snap to layout" would snap P1, P2 or both to layout edges.

The seed points would be P1 and P2 then. If you set P1 = P2, the effect is the same as for "create_measure_ruler". All these buttons imply "Apply".

You can try this solution in the "issue-342" branch.

Best regards,

Matthias

Kazzz-S, 2019-09-01:
Hi Matthias,

Thank you again for your positive and prompt action.
Your idea is fine with me. "Swap points" is the one I want, too!

I'll try the branch.

With warm regards,
Kazzz

Kazzz-S, 2019-09-03:
Hi Matthias,

I've confirmed that the new GUI in the "issue-342" branch works as you described.
It's fascinating and makes my job easy!

With best regards,
Kazzz

closed 2019-09-03 by Kazzz-S