KLayout/klayout #343: Feature request (low priority): Bookmarks in "Dock Widget"

Kazzz-S, 2019-09-03:
While testing #341 and #342, I needed a bit long bookmark list like:

[image]

It would be convenient if the bookmarks are shown in a **Doc Widget** (or pinned container) not to follow the menu hierarchy again and again.

With warm regards,
Kazzz

klayoutmatthias, 2019-09-08:
This is how it looks:

[image]

Kazzz-S, 2019-09-08:
Awesome!!!

Thank you so much again.

With warm regards,
Kazzz

klayoutmatthias, 2019-09-11:
It's released :-) 

Ok to close this?

BR, Matthias

Kazzz-S, 2019-09-11:
Hi Matthias,

I've built 0.26 on Linux (my primary environment) and found this new feature is handy as expected.
Once again, thank you very much for providing this feature.
Yes, please close this ticket.

I've demonstrated the new version, and my colleagues love this feature, too, and conceived a new, improved workflow using this feature.

One commented that it would be much better if a bookmark can be selected not only by the mouse click but also by the arrow keys if the focus is in the dock widget.

With warm regards,
Kazzz

Kazzz-S, 2019-09-11:
Hi Matthias,

What it should have been:
> One commented that it would be much better if a bookmark can be selected not only by the mouse click but also by the arrow keys if the focus is in the dock widget.

One commented that it would be much better if item selection/change by the arrow keys takes the same effect as the mouse click.

Regards,
Kazzz

klayoutmatthias, 2019-09-12:
Hi kazzzthis is there already :-)Enable "Follow selection" in the right-click menu of the bookmark widget. Then the bookmark will be updated when the selection is changed with the arrow keys.Matthias

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/KLayout/klayout/issues/343?email_source=notifications\u0026email_token=AGEHQCMZIG3TYI7FTMQGEQ3QJF4WVA5CNFSM4ITEGP62YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD6QFXFQ#issuecomment-530602902",
"url": "https://github.com/KLayout/klayout/issues/343?email_source=notifications\u0026email_token=AGEHQCMZIG3TYI7FTMQGEQ3QJF4WVA5CNFSM4ITEGP62YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOD6QFXFQ#issuecomment-530602902",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]