KLayout/klayout #346: Unexpected cell bbox calculation

dimapu, 2019-09-04:
I would like to thank the authors for the great software.

I have faced a strange result when checking the bbox of a cell containing another cell. See for example the drawing: 
[image]

The top cell's bounding box appears to be larger than the shape: it also includes the whole bounding box of the cell 1. The bounding box for the layer containing the shape also includes cell 1. From the documentation for `bbox_per_layer()`: "The bounding box is the box enclosing all shapes on the given layer. "

The code to generate this:
```python
from klayout.dbcore import Layout, Cell, Polygon, Point, CellInstArray
from klayout.dbcore import ICplxTrans, LayerInfo

# Create layout
l = Layout()

# Add Cell 1
l.add_cell('1')
cell = l.cell('1')
print(cell.bbox())  # ()

# Add a rotated by 45 degree polygon into Cell 1 into a new layer
layer = l.layer(LayerInfo(1, 0))
poly = Polygon([
    Point(500, 0),
    Point(0, 500),
    Point(500, 1000),
    Point(1000, 500),
])
cell.shapes(layer).insert(poly)

print(cell.shapes(layer).size())  # 1
print(cell.bbox())  # (0,0;1000,1000)

# Create a top cell
l.add_cell('top')
top_cell = l.cell('top')
print(top_cell.bbox())

# Rotate cell 1 by 45 degrees and add it into the top cell
t = ICplxTrans(1, 45, False, 0, 0)
print(t)  # r45 *1 0,0
cia = CellInstArray(cell.cell_index(), t)
print(cia)  # #0 r45 *1 0,0
top_cell.insert(cia)

# Check the bbox of the top cell
print(top_cell.bbox())  # (-707,0;707,1414)
print(top_cell.bbox_per_layer(layer))  # (-707,0;707,1414)
print(top_cell.called_cells())

# Check the bbox of the top cell after flattening
top_cell.flatten(False)
print(top_cell.shapes(layer).size())  # 1
print(top_cell.bbox())  # (-354,354;354,1061)
print(top_cell.bbox_per_layer(layer))  # (-354,354;354,1061)
```
There would be two ways to resolve it:
1. Fix `bbox` method so that it includes only actual shapes. OR
2. Fix documentation of the `bbox` and `bbox_per_layer` methods to reflect the current behavior. 

In the latter case, I would like to ask a feature request to have `bbox_shapes_only` and `bbox_per_layer_shapes_only` methods which calculate the bbox based on the shapes only. It can now be done by first flattening the cell and then returning the bbox value (see the last lines in the code), but may be there is a more efficient way.

klayoutmatthias, 2019-09-04:
Well ... the documentation doesn't say, the bounding box is the "smallest box" :-)

That's not just a joke: the bounding box is an *outer* approximation of the cell's area and used for example to identify cells that need to be visited in a region query. We don't want to miss a cell, that's why a cell's bounding box must not be smaller than the cell. But it can be bigger which just renders a small performance penalty in this application.

In your case the issue is the cell's rotation. A cell's bounding box is computed by the bounding boxes of all shapes and all child cell instances. A child cell instance's bounding box is given by the bounding box of the transformed bounding box of the cell. The diamond has a bounding box which is bigger, the transformed bounding box will be a diamond again and the bounding box of that diamond is even bigger. But this scheme allows bottom-up, cached bounding box computation and is very fast.

To compute the elaborate version of bounding box you ask for, the bounding box needs to be computed by analysing the shapes inside a cell under the influence of all imposed transformations. Only this way you can detect this case, when the diamond gets rotated into a box. This eventually is nothing more than flattening and the only opportunity for optimisation is to keep a cache of transformation variants. That's not feasible in the framework of the current bounding box computation.

Regards,

Matthias

P.S. nice issue report BTW!

dimapu, 2019-09-05:
Dear Matthias, 

Thank you for the quick response! You are right about the "smallest box"! :) It indeed makes sense to calculate the bounding box the way you described in terms of speed. 

I am actually using more complex cells than shown in the example. And I was misusing the `bbox()` method to find the centers and the sizes of my cells, and later positioning them adjacent to each other, hoping that the layers inside will be touching. I will switch to first flattening the cells and taking the bbox after that.

May be you would still consider adding few lines to the method documentation to explicitly explaining the way it is calculated. I could submit a PR with this (file gsiDeclDbCell.cc, right?). Otherwise I will close this issue.

Best regards, 
Dima.

closed 2019-09-05 by dimapu