KLayout/klayout #353: file-path and rdb-file behavior is inconsistent

EugeneZelenko, 2019-09-12:
When session is loaded with -u layout files as loaded from same directory as session file, but rdb-file is loaded from current working directory. This leads to problem when path in -u and current working directory are different.

Workaround is to use absolute or relative path in rdb-file, but I think will be good idea if KLayout behavior will be consistent in both cases.

klayoutmatthias, 2019-09-13:
Good point. Makes sense.

Matthias

EugeneZelenko, 2019-09-13:
Sorry, I forgot to mention that I meant that both file-path and rdb-file are just file names without path.

klayoutmatthias, 2019-09-14:
I also noticed that L2N and LVS DB are not included in the sessions. This is inconsistent as RDB's are. I have fixed this as well.

EugeneZelenko, 2019-09-14:
Thank you for quick fix!

commit 6a305cfb (2019-09-16): Merge pull request #356 from KLayout/issue-353

Fixed #353 (paths relative to .lys file for rdb-file)
commit b747bbab (2019-09-16): Fixed #353 (paths relative to .lys file for rdb-file)

This fix also includes: L2N and LVS DB files are now also
included in the sessions.
closed 2019-09-16 by klayoutmatthias