KLayout/klayout #359: Feature request: PNG screenshot with even-number pixels

Kazzz-S, 2019-09-17:
I quite often use the **Screenshot** and generate a movie file (MP4) from captured PNG files using **"FFmpeg"** for presentation purpose.
In this process, the PNG files should have the size of even-number pixels by even-number pixels.

It would be convenient if the **Screenshot** has an option to adjust the image size forcibly to the nearest even-number pixels.

Thank you and best regards,
Kazzz

klayoutmatthias, 2019-09-17:
Hi Kazzz,

there is an old script which provides a screenshot with options: https://www.klayout.de/useful_scripts.html#screenshot.lym

The basic idea is a simple Ruby one-liner:

```
RBA::LayoutView::current.save_image_with_options("x.png", 800, 600, 0, 0, 0, RBA::DBox::new, false)
```

The first argument is the file name, the second and third the width and height of the image. The other values are defaults for the additional parameters supported by this method.

Best regards,

Matthias

Kazzz-S, 2019-09-17:
Hi Matthias,

Thanks for the suggestion.
I should have checked the resources first.
The Ruby one-liner is sufficient enough.
Let me close this ticket.

With warm regards,
Kazzz

closed 2019-09-17 by Kazzz-S