KLayout/klayout #360: Building 0.26 for macOS

Kazzz-S, 2019-09-17:
Hi Matthias,

I'm trying to build the new major release **0.26** (thanks a lot for this release!) for macOS, too.
The first issue is a link error in making **libdb_tests***.

A build log file (with **make -j1** option) is attached below.
[qt5.build.macos-Yosemite-release.01.zip](https://github.com/KLayout/klayout/files/3622005/qt5.build.macos-Yosemite-release.01.zip)
Please kindly investigate this case.

With warm regards,
Kazzz

klayoutmatthias, 2019-09-17:
Hi Kazzz,

I tried to figure out the problem, but I can't find an obvious reason. I'm using clang myself, but I have not seen this problem.

I have difficulties running MacOS in VirtualBox right now, but I'll try later. Maybe I'm able to reproduce it then.

BTW: Did you try a complete rebuild from a clean work area?

Best regards,

Matthias

Kazzz-S, 2019-09-17:
Hi Matthias,

Thanks for your response.
> I tried to figure out the problem, but I can't find an obvious reason. I'm using clang myself, but I have not seen this problem.
> 
> I have difficulties running MacOS in VirtualBox right now, but I'll try later. Maybe I'm able to reproduce it then.

Please take time.

> BTW: Did you try a complete rebuild from a clean work area?

Yes. The exiting target "build*/" and "bin*/" directories were completely deleted before starting.

With warm regards,
Kazzz

klayoutmatthias, 2019-09-17:
Hi Kazzz,

I'm running a fresh build on 0.26 with my VirtualBox instance of MacOS and it looks like it's not showing this error :-(

This is my configuration:

Apply LLVM version 9.0.0 (clang-900.0.39.2)
Target: x86_64-apple-darwin17.0.0
Qt 5.9.3

I had to disable Python for now, but I don't think this is responsible for this issue.

Matthias

klayoutmatthias, 2019-09-17:
I should add that my build is running of master branch, but it should be more or less identical to the 0.26 release.

Matthias

klayoutmatthias, 2019-09-18:
It succeeded for me :-(

Maybe there is a difference in the compiler in your case? I imaging the problem might occur if the compiler would not export explicit template instantiations. 

I can pinpoint the origin: it's probably related to the explicit template instantiation of local_processor in line 1617 of dbHierProcessor.cc. As this instantiation is marked "__attribute__ ((visibility ("default")))" the compiler is supposed to export the symbols of this instantiation. This seems to fail for the non-inlined functions such as "local_processor<TS, TI, TR>::compute_local_cell(...)" etc. 

Those methods are not marked as "to export", but this does not seem to be required in my cases. But it also does not seem to hurt. 

I have tried to put as many of those export statements (aka "DB_PUBLIC") into the code and pushed this to "macos-fix-maybe" branch (CAUTION: this isn't 0.26 but *master*). Maybe you can check if it works with these changes (or: works better :-))

Matthias

Kazzz-S, 2019-09-18:
Hi Matthias,

Thank you for your effort in solving this issue.
I'll try "macos-fix-maybe" branch and get back to you.

Meanwhile, I'm trying to build the "master" branch on different OS. This is an intermediate status.
### Yosemite 10.10.5 ###
```
Apple LLVM version 7.0.2 (clang-700.1.81)
Target: x86_64-apple-darwin14.5.0
Thread model: posix
```
**This is the case of this ticket.**

### El Capitan  10.11.6 ###
```
Apple LLVM version 8.0.0 (clang-800.0.42.1)
Target: x86_64-apple-darwin15.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```
**This succeeded!**
[image]

### High Sierra 10.13.x (your environment)  I'll also try later. ###
```
Apple LLVM version 9.0.0 (clang-900.0.39.2)
Target: x86_64-apple-darwin17.0.0
```
**This succeeded!**

Looking in tho these results, I think this issue is basically "old compiler issue."

With warm regards,
Kazzz

Kazzz-S, 2019-09-25:
Hi Matthias,

Here is the status as of today.
I could successfully build the "**master**" branch on four different OS out of five, as shown above.
Only the oldest **Yosemite** environment is problematic;-(
With some reasons, I cannot say goodbye to **Yosemite** right now, which is my personal issue.
 
> I have tried to put as many of those export statements (aka "DB_PUBLIC") into the code and pushed this to "macos-fix-maybe" branch (CAUTION: this isn't 0.26 but _master_). Maybe you can check if it works with these changes (or: works better :-))

I've tried "**macos-fix-maybe**" branch on **Yosemite**, however, the problems remain unchanged completely, which is weird.  [qt5.build.macos-Yosemite-release.03.zip](https://github.com/KLayout/klayout/files/3654849/qt5.build.macos-Yosemite-release.03.zip)

I'm Googling to find some clues but not found yet. 
If you conceive of anything, please let me know.

With warm regards,
Kazzz

matthewvanniekerk, 2019-10-11:
Hello guys - I wasn’t sure if you would prefer a new topic or contribution to this one. (I can move the question if needed)
I have recently (perhaps pre-maturely) upgraded to the MacOS 10.15 Catalina and to my disappointment KLayout is no longer able to function. Are there plans to include this OS in the future Mac release? I’ve had some trouble getting the Mac build scripts to work on my machine, it seems that the changes made have implicated some permissions details and things like XCode.

My limited tinkering with the scripts have yielded a failure to create the bin directory in the build, but if anything changes or I find a way to make it work I’ll pass it along.. 

Thanks!

Kazzz-S, 2019-10-12:
Hi @matthewvanniekerk,

Thank you for mentioning about **Catalina** (macOS 10.15) case.
Since I'm not in a position to have **Catalina** for the time being, your attempt is a good case study.

With warm regards,
Kazzz

matthewvanniekerk, 2019-10-21:
@Kazzz-S from what I can tell, everything is working well except for the trace net. However, my build was based on qt5 brew and python brew, which were marked as experimental. I get a strange C error when I try to run the trace all nets.

If you want, I can provide more details.

Matt

Kazzz-S, 2019-10-21:
Hi Matt,

Thank you for your comment.
So far, I've not tested the net tracing features of 0.26, which is a major release, on any OS.
If you found any problems regarding the features, I think, it would be better to issue a separate ticket since the problems may reproduce on different platforms, too and your problem may not be related to the **experimental** environment.

----
Comments on **experimental** environments:
Earlier (~KLayout 0.23.2 or so), Qt was a part of Mac OSX and KLayout was built using those OS-bundled Frameworks including Ruby.
I was one of the users of such binary packages provided on the community site.

Later, Qt was dropped from Mac OSX.
When KLayout 0.25 (with Python support) was released, I tried to build it by myself (#20) using:
 1. Qt from MacPorts because I was a bit familiar with the tool kit; no other reasons
 2. OS-bundled Ruby and Python to minimize dependency on external libraries

which was assumed to be the **standard**.

Of course, some people like Homebrew (including you) and others may like Anaconda3 (https://github.com/Kazzz-S/klayout/issues/34).
So, these environments are just marked **experimental** in contrast with the **standard**.

I'm planning to update the build environment for 0.26 as I did for 0.25.

Regards,
Kazzz

atait, 2019-10-28:
I have been modifying build4mac.py for Catalina. The challenge is the new ruby bundled with MacOSX. If you don't need ruby (like me), you can use these DMGs of 0.26 compiled on Catalina.
https://www.dropbox.com/sh/40myqi8brunje6s/AACW3_iVJiTq7zHI8mEOT9Xka?dl=0

@Kazzz-S Here are some possibly useless notes if you want to give it a shot. I tried these installations of ruby:
- Catalina built-in (/System/Library/Frameworks/Ruby.framework/Versions/2.6)
    - It no longer comes with the Headers directory, which contains the .h files
    - That was the main issue beyond simply changing the version number in build4mac_env.py
- Xcode (/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/Ruby.framework/Versions/2.6)
    - Does not come with a ruby executable
- rbenv (~/.rbenv/versions/2.6.4)
    - Does not have a libruby.dylib. Instead it is libruby.2.6-static.a. That should work, but it gets linker errors
- Homebrew (/usr/local/Cellar/ruby/2.6.5)
    - This has all the necessary components, and building works
    - The resulting application's icon bounces in the dock, then closes

I also tried mixing and matching the Headers from Homebrew with executable and dylib from Catalina, without success.

The Homebrew approach should have worked, but perhaps klayout 0.26 is just not compatible with ruby 2.6. I tried installing ruby 2.3 through rbenv and got a cryptic "ERROR: Ruby install aborted due to missing extensions".

@matthewvanniekerk Were you able to get around the issue of the missing Headers? Why did you downgrade Xcode - I've not seen any issues with Qt.

Best,
Alex

Kazzz-S, 2019-10-28:
Hi @atait,

Thank you for your effort and information about Ruby on **Catalina**. In my case, Ruby is a must:-)
I faced a similar issue (missing *.h) in **Mojave** and had to wait for the release of a **beta Xcode for developers**. Let's see what happens in the future.

I'm going to set up Catalina OS and update the environment for 0.26 in the near future.

Regards,
Kazzz

matthewvanniekerk, 2019-10-28:
@atait Thanks for the advice! I was able to compile with the Homebrew version of ruby (even though I don't use it, like you) and had no issue with the headers. I pointed the previous "header" directory to '/usr/local/Cellar/ruby/2.6.5/include/ruby-2.6.0' in the build4mac_env.py file, where I found the ruby.h file. I didn't really question this because it just worked.

As for downgrading Xcode, I think I was a bit early on this build and Qt may not have been released for the new version. I didn't think this would be a permanent issue I just wanted to note what I did to make mine work.

Matt

Kazzz-S, 2019-12-15:
**Catalina 10.15.2: (Qt, Ruby, Python)=(Macports, OS-bundled, OS-bundled)**

I finally gave up to support **Yosemite** and purchased a new Mac after many years!

```
Apple clang version 11.0.0 (clang-1100.0.33.16)
Target: x86_64-apple-darwin19.2.0
Thread model: posix
```
[image]
[image]

Works are still going, but the intermediate product **klayout-0.26.1-macOS-Catalina-1-qt5MP-RsysPsys.dmg** is available here:
https://www.dropbox.com/sh/9tr90u3fq47aaql/AADMghnjJRp20fGkhUNyAseWa?dl=0

Regards,
Kazzz

Kazzz-S, 2019-12-15:
**Catalina 10.15.2: (Qt, Ruby, Python)=(Anaconda3, Anaconda3, Anaconda3)**

```
Apple clang version 11.0.0 (clang-1100.0.33.16)
Target: x86_64-apple-darwin19.2.0
Thread model: posix
```

[image]
[image]

This intermediate product **klayout-0.26.1-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg** is also available under https://www.dropbox.com/sh/9tr90u3fq47aaql/AADMghnjJRp20fGkhUNyAseWa?dl=0

Please note that this is a **_half-cooked package_** for those who have the latest **anaconda3** installed under **$HOME/opt/anaconda3**. 
Qt Framework is not included in the DMG. Therefore, the DMG file size is small.
You also need to install Ruby by:
```
$ conda install ruby
```
To invoke this application bundle, you need to set **PYTHONHOME** environment variable properly:
```
export PYTHONHOME=$HOME/opt/anaconda3
```

A simple way is to use **KLayoutAna3.app** script bundle included in **script-bundles.zip**.
Of course, you and make your script bundle using **Automator.app**.

Regards,
Kazzz

Kazzz-S, 2019-12-22:
Let me close this thicket since I have start a new one: #457 

Regards,
Kazzz

closed 2019-12-22 by Kazzz-S