KLayout/klayout #379: Dividing error (?) when saving long PATHs to GDS2

Kazzz-S, 2019-10-12:
Please refer to the attached ZIP file [long-path.zip](https://github.com/KLayout/klayout/files/3720295/long-path.zip).

1. **curl.oas** is an original design file in OASIS containing a long PATH.
2. **curl-KL.gds** is generated by KLayout with "save as" command.
3. **curl-XX.gds** is anther GDS2 file converted by a commercial tool.

When opening **curl-KL.gds**, I got the error shown below.
[image]

Other conditions are as follows.
[image]
[image]

If I enable the **Multi-XY record mode**, KLayout can open the generated GDS2 normally.
However, some CAE tools require backward compatibility (old style).

Thank you and best regards,
Kazzz

klayoutmatthias, 2019-11-01:
Hi Kazzz,

you're right - the problem is that for paths - unlike polygons - there is no good "split" approach. The problem is manifold: 
* Path ends of the split path might not combine to a full path
* Rounded path ends cannot be specified for one end only
* The interactions between path segments can be very complex in the general case, so splitting might result in different outlines

What I can do however is to convert paths with too many points into polygons and apply splitting there.

Would this be a feasible solution?

Best regards,

Matthias

Kazzz-S, 2019-11-01:
Hi Matthias,

Thanks for your reply.

> What I can do, however, is to convert paths with too many points into polygons and apply splitting there.
> 
> Would this be a feasible solution?

Yes, I think this can be a solution. 
The various issues regarding long PATHs you described above are also important.
So when such a PATH is found in saving to GDS2, I think, it would be better to flag on a warning to
prompt the user to select (1) using Multi-XY record mode or (2) converting it to a set of polygons.
Or such a choice can be added to the **Writer Option** from the beginning.

Fortunately, in my case, I can use a commercial tool, and its "split approach" shown in the attached file works okay with the CAE tool. Moreover, the CAE tool can accept POLYLINEs in DXF.
Both the data were computed normally, and the results agreed perfectly. GDS2 was much faster, though:-)

Warm regards,
Kazzz

stefanottili, 2025-05-08:
It seems that nobody is running into a problem with path > 8191 points.
This issue has been open and unsolved for six years ... ;-)