KLayout/klayout #381: KLayout does not work on MacOS Catalina

garethsion, 2019-10-14:
Hi there,

I've just updated my MacBook OS to the new Catalina. However, upon doing so, I've found that Layout no longer works, and there doesn't seem to be a version which will work. Are there any plans to roll KLayout out for Catalina in the immediate future / is there any way I can ameliorate this?

Thanks
G

Kazzz-S, 2019-10-14:
Hi,

I think the main reason is the difference in the Ruby versions: expected (2.3) and actual (2.6).
Please refer to https://github.com/KLayout/klayout/issues/297. If you do not need Ruby, a special binary package https://www.dropbox.com/sh/qeny6se8fepgois/AACnthqtYTGxN78XhF02rav3a?dl=0 will work.

However, this is version 0.25.9, not the latest 0.26.
The status for 0.26 is here (https://github.com/KLayout/klayout/issues/360).
To get KLayout 0.26 for Catalina, we need to build it on Catalina itself.
Unfortunately, I'm not in a position to use Catalina for the time being;-(.
 
Thanks
Kazzz

garethsion, 2019-10-15:
Hi Kazzz, 

Thank you very much for your reply. I've downloaded the binary you attached, and so far it seems to work well on the new OS. I'll run through and make sure it all works properly, but so far so good. Thanks again - much appreciated.

G

thbesson, 2019-10-18:
Thanks for the package without ruby, it works perfectly with catalina :) Life saver!

atait, 2019-10-28:
I built 0.26 on Catalina. No ruby. The DMGs are here
https://www.dropbox.com/sh/40myqi8brunje6s/AACW3_iVJiTq7zHI8mEOT9Xka?dl=0

One of them has system python 2.7. The other has homebrew python 3.7, which I need. As far as I know, there hasn't been a public python3 binary since 0.25.3, so I hope that is helpful for someone.

If you are interested in the compile details, I put some notes in #360 

Best,
Alex

thbesson, 2019-10-28:
Thanks, it seems to work really well.

lukasc-ubc, 2019-10-28:
Hi Alex,
Thank you very much for working on this.  Many things are working well, but I am having issues with Qt functionalities in SiEPIC-Tools.  
Which version of Qt are you using?  (And which one was used by KLayout in the previous versions).

Here is an example error (which didn’t happen in 0.25.9)

> wdg.setAttribute(pya.Qt.WA_DeleteOnClose)
Unexpected object type (expected argument of class EmptyClass, got Qt_WidgetAttribute) in QWidget_Native.setAttribute
(eval):1

Thank you
Lukas

atait, 2019-10-29:
Hi Lukas,

This one has Qt 5.11.1 and the other one I have is using had Qt 5.10.1. I can get your error in the new one, and I don't get it in the old one. That is likely the issue. The latest stable release of Qt on brew is 5.13.1. Perhaps they patched this incompatibility. I will rebuild and get back to you.

Also, I encountered an odd thing with SiEPIC-Tools. It does not find PCellDeclarationHelper via a star import. I think the issue is it not being added to pya's `__all__` attribute. To fix it, I added this line

```
from pya import PCellDeclarationHelper
```

after the below line in "SiEPIC - General Library.lym". I don't know why that isn't a problem for everyone or in 0.25
https://github.com/lukasc-ubc/SiEPIC-Tools/blob/f5b5f14c57217c20695a4ffd6503cd7f7d331b9b/klayout_dot_config/pymacros/SiEPIC%20-%20General%20Library.lym#L25

lukasc-ubc, 2019-10-29:
Thanks Alex. I’m glad you can see the same error. 

I am fixing the libraries to use pya. PCellDeclarationHelper rather than just PCellDeclarationHelper

atait, 2019-10-29:
Building with Qt 5.13.1 did not solve the problem unfortunately. I cannot figure out how to downgrade to Qt 5.10.1 through brew. In #360, Matt was also able to compile on Catalina. I think he uses SiEPIC-Tools. You'd have to ask him whether it works for him. In either case, it might become necessary to update SiEPIC-Tools for the latest Qt. Hopefully that will be a matter of googling and then find-all/replace.
-Alex

garethsion, 2019-10-29:
Thanks very much for working on this Alex.

lukasc-ubc, 2019-11-02:
Hi Alex,

In your 0.26 OSX build, I get the following errors when running an empty py file from the command line:

lukascMacBook13:GitHub lukasc$ mkdir tmp
lukascMacBook13:GitHub lukasc$ cd tmp
lukascMacBook13:tmp lukasc$ touch test.py
lukascMacBook13:tmp lukasc$ /Applications/klayout\ 0.26.app/Contents/MacOS/klayout -b test.py
Warning: Unable to find GSI class binding for: N2Qt15HitTestAccuracyE
Warning: Unable to find GSI class binding for: 5QPairIiiE
Warning: Unable to find GSI class binding for: N2Qt16ApplicationStateE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt19ImageConversionFlagEE
Warning: Unable to find GSI class binding for: N2Qt10BrushStyleE
Warning: Unable to find GSI class binding for: N2Qt11GlobalColorE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt16KeyboardModifierEE
Warning: Unable to find GSI class binding for: N2Qt11CursorShapeE
Warning: Unable to find GSI class binding for: N2Qt10DropActionE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt10DropActionEE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt11MouseButtonEE
Warning: Unable to find GSI class binding for: 6QFlagsIN9QIODevice12OpenModeFlagEE
Warning: Unable to find GSI class binding for: N2Qt11FocusReasonE
Warning: Unable to find GSI class binding for: N2Qt13TextElideModeE
Warning: Unable to find GSI class binding for: N2Qt15LayoutDirectionE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt13AlignmentFlagEE
Warning: Unable to find GSI class binding for: N2Qt8MaskModeE
Warning: Unable to find GSI class binding for: N2Qt15AspectRatioModeE
Warning: Unable to find GSI class binding for: N2Qt18TransformationModeE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt16InputMethodQueryEE
Warning: Unable to find GSI class binding for: N2Qt16InputMethodQueryE
Warning: Unable to find GSI class binding for: N2Qt11MouseButtonE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt14MouseEventFlagEE
Warning: Unable to find GSI class binding for: N2Qt16MouseEventSourceE
Warning: Unable to find GSI class binding for: N2Qt17NativeGestureTypeE
Warning: Unable to find GSI class binding for: 9QMarginsF
Warning: Unable to find GSI class binding for: 8QMargins
Warning: Unable to find GSI class binding for: N2Qt17ScreenOrientationE
Warning: Unable to find GSI class binding for: N2Qt14WindowModalityE
Warning: Unable to find GSI class binding for: N2Qt11WindowStateE
Warning: Unable to find GSI class binding for: N2Qt6BGModeE
Warning: Unable to find GSI class binding for: N2Qt13ClipOperationE
Warning: Unable to find GSI class binding for: N2Qt8FillRuleE
Warning: Unable to find GSI class binding for: N2Qt8SizeModeE
Warning: Unable to find GSI class binding for: N2Qt8PenStyleE
Warning: Unable to find GSI class binding for: N2Qt11PenCapStyleE
Warning: Unable to find GSI class binding for: N2Qt12PenJoinStyleE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt17ScreenOrientationEE
Warning: Unable to find GSI class binding for: N2Qt10CheckStateE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt8ItemFlagEE
Warning: Unable to find GSI class binding for: N2Qt9SortOrderE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt9MatchFlagEE
Warning: Unable to find GSI class binding for: N2Qt11OrientationE
Warning: Unable to find GSI class binding for: N2Qt10TextFormatE
Warning: Unable to find GSI class binding for: N2Qt16TabFocusBehaviorE
Warning: Unable to find GSI class binding for: N2Qt15CursorMoveStyleE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt15TouchPointStateEE
Warning: Unable to find GSI class binding for: N2Qt15TouchPointStateE
Warning: Unable to find GSI class binding for: N2Qt4AxisE
Warning: Unable to find GSI class binding for: N2Qt11ScrollPhaseE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt10WindowTypeEE
Warning: Unable to find GSI class binding for: N2Qt10WindowTypeE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt11WindowStateEE
Warning: Unable to find GSI class binding for: N2Qt16InputMethodQueryE
Warning: Unable to find GSI class binding for: N2Qt17ItemSelectionModeE
Warning: Unable to find GSI class binding for: N2Qt13TextElideModeE
Warning: Unable to find GSI class binding for: N2Qt10DropActionE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt16KeyboardModifierEE
Warning: Unable to find GSI class binding for: 6QFlagsIN19QItemSelectionModel13SelectionFlagEE
Warning: Unable to find GSI class binding for: 8QMargins
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt10DropActionEE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt13AlignmentFlagEE
[... 61 lines cut ...]
Warning: Unable to find GSI class binding for: 6QFlagsIN9QIODevice12OpenModeFlagEE
Warning: Unable to find GSI class binding for: 5QPairI10QByteArrayS0_E
Warning: Unable to find GSI class binding for: N2Qt11OrientationE
Warning: Unable to find GSI class binding for: N2Qt10DropActionE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt8ItemFlagEE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt9MatchFlagEE
Warning: Unable to find GSI class binding for: N2Qt9SortOrderE
Warning: Unable to find GSI class binding for: 6QFlagsIN2Qt10DropActionEE
Warning: Unable to find GSI class binding for: N2Qt15AspectRatioModeE
Warning: Unable to find GSI class binding for: N2Qt17ItemSelectionModeE
Warning: Unable to find GSI class binding for: N2Qt16InputMethodQueryE
[... cut ...]
Warning: Unable to find GSI class binding for: N2Qt16InputMethodQueryE
Warning: Unable to find GSI class binding for: 9QtMsgType
ERROR: Can't run macro (no interpreter): :/built-in-macros/pcell_declaration_helper.lym
ERROR: Can't run macro (no interpreter): :/built-in-macros/qobject_helper.lym
lukascMacBook13:tmp lukasc$ /Applications/klayout\ 0.25.9.app/Contents/MacOS/klayout -b test.py
klayout 0.25.9.app/
lukascMacBook13:tmp lukasc$ /Applications/klayout\ 0.25.9.app/Contents/MacOS/klayout -b test.py
lukascMacBook13:tmp lukasc$

atait, 2019-11-02:
I am getting the same warnings. They do not seem to affect my code like they do to SiEPIC code. Starting a new thread in SiEPIC
https://github.com/lukasc-ubc/SiEPIC-Tools/issues/103

klayoutmatthias, 2020-06-29:
This topic has been addressed and fixed in #589. Please note, that on MacOS certain build variants are not working reliably. Those are marked as "experimental" in the downloads section.

closed 2020-06-29 by klayoutmatthias