KLayout/klayout #382: Possibility to copy user property value

EugeneZelenko, 2019-10-15:
It'll be great to have possibility (via context menu/keyboard shortcut) to copy values from User Properties dialog executed from Object Properties dialog.

klayoutmatthias, 2019-11-02:
Hi Eugene,

I have extended the user properties dialogue, so alternatively you can edit the properties with a text editor. This is not just faster but also allows copy and paste because it's plain text. And it has syntax highlighting :)

[image]

Matthias

EugeneZelenko, 2019-11-02:
Hi, Matthias!

Thank you for so fast implementation! Sure, your design is much more flexible and convenient. I just concentrated on immediate need :-)

klayoutmatthias, 2019-11-13:
Will be released soon with 0.26.1

closed 2019-11-13 by klayoutmatthias