KLayout/klayout #384: Strange behavior of Try-execept block in Python scripts

jurask, 2019-10-22:
When exception is rised inside python script and Macro development panel is opened an error window appears even when the exception is properly handled

Considering the following script

```
import pya, time

# Enter your Python code here ..

p = RelativeProgress("qaaa", 100)
for i in range(100):
    try:
        time.sleep(1)
        p.inc()
    except RuntimeError as e:
 #   except:
        print(e)
        break
p.destroy()
```

when user clicks Cancel button an RuntimeError exception is thrown which is handeled by try-catch block, unfortunatelly when the macro development panel is opened an error window appears stating there was an error interrupting normal code execution and error handling

klayoutmatthias, 2019-11-02:
Hello,

I'm afraid that's intrinsic.

The macro IDE catches exception when they are *thrown*, because this is usually the place where you want to debug. "Cancel" from the progress bar is implemented through an exception. So when you press cancel, the exception is shown in the macro IDE.

There are two ways to mitigate this:
* You disable "Stop on exception" in the Macro Editor Options (see screenshot below)
* I implement a special handling for CancelException so it's not shown.

Matthias

[image]

klayoutmatthias, 2019-11-02:
I noticed one important thing with your example: If you stop the script in the breakpoint, the main window is blocked. 

The reason for this is the following: as long as the RelativeProgress object is alive, it will capture the main window. This is a built-in behavior of the RelativeProgress object. Normally that's not an issue because Python cleans up objects when leaving the scope (this will actually call "destroy"). In your case this does not happen, because "p" is defined on top level, hence it's a global variable which never goes out of scope.

This version of the script avoids this by using a function scope. It does not even need ```destroy()```:

```
import pya
import time

# Enter your Python code here ..

def func():

  p = pya.RelativeProgress("qaaa", 100)
  for i in range(100):
      try:
          time.sleep(1)
          p.inc()
      except RuntimeError as e:
          print(e)
          break

func()
```

Matthias

klayoutmatthias, 2019-11-13:
Will be released soon with 0.26.1

closed 2019-11-13 by klayoutmatthias