KLayout/klayout #387: Salt, Python version

lukasc-ubc, 2019-11-01:
Hi Matthias,

In the Salt grain.xml, https://www.klayout.de/package_cookbook.html, you provide the ability to provide a minimum version for dependencies. It is very useful. 

If I have a package that requires a minimum Python version, is it possible to add this?  Specifically, we have KLayout builds for Python 2.7 and 3.6 for Mac OS, and I would like be able to check if the user has the right one.  If the user has the wrong one, then ideally KLayout would display a message with a URL link to the correct download page.

If this is not available, could you please add Python version as an option in grain.xml?  

Thank you
Lukas

klayoutmatthias, 2019-11-03:
The proposal I have is to encode it in "api-version". This is a kind of natural extension.

So:
* api-version = "0.26" would check for KLayout 0.26 or later
* api-version = "python" would check whether Python is included
* api-version = "python 3.6.1" would check whether Python is included and 3.6.1 or later
* api-version = "0.26; python 3.6.1" would check all of that

In addition, more features can be added, e.g.:
* api-version = "0.26; python 3.6.1; qt_binding" would also check if Qt bindings have been compiled in

This scheme is backward compatible as older version just check for the initial part. However, old versions would not complain.

Right now, there is a warning in the package selector:

[image]

But you can still install the package if you really want to.

How about this? This implementation is available as PR #396.

Matthias

thomaslima, 2019-11-03:
Dear Matthias. Thank you for this. Two questions:

1. I think we should be more explicit and use qualifiers >, >=, ==, <=, < for version number.
2. Does 3.6 mean 3.6.*?

lukasc-ubc, 2019-11-03:
This is excellent! Thank you.

klayoutmatthias, 2019-11-03:
Hi Thomas,

after your comment I have included pseudo-features, specifically "python3" and "python2" which allow selecting for a specific Python major version. So if you want Python 2 only, you can specify:

```
<api-version>python2</api-version>
```

Selecting Python 2 with 2.7 at least is possible with:

```
<api-version>python2 2.7</api-version>
```

and simply saying 

```
<api-version>python3</api-version>
```

will give you any Python 3 version, but not Python 4, whatever this will bring ...

Matthias

thomaslima, 2019-11-03:
Understood. I think this is a great addition. Thank you for working on this.

klayoutmatthias, 2019-11-13:
Will be released soon with 0.26.1

closed 2019-11-13 by klayoutmatthias