KLayout/klayout #401: Feature: support more xsection script entries, and possibly manage list of entries

cfriesicke, 2019-11-05:
Currently, after being opened/executed, xsection scripts are added to a list of history entries that is typically stored in $HOME/.klayout-xsection (with XML- or XML-like syntax). The number of entries in that list is **limited to four entries**.

When using more than four different xsection scripts, the use or xsection scripts can become a bit inconvenient: the history entries will always rotate, depending on which script was called last, and those scripts that are not in the history list have to be opened again using the file dialog. It would be nice if the history list could be longer than just four entries; ten entries seems reasonable in my opinion. I guess that this could be really easy to implement, maybe it's simply a constant value that needs to be changed?

Ultimately, it would be nice if xsection scripts could be added to a list of "Favorites" but that would require some sort of GUI dialog where they can be arranged in a certain order. IMO this is only "nice to have" and not really essential. Having more history entries would already be sufficient.

Background:
We use KLayout _a lot_ where I work (III-V technology development and MMIC design) since it runs very fast, works on all platforms, and since the scripting APIs are well-documented (and not behind paywalls...). We maintain several semiconductor technologies and like to use xsection there every now and then to draft ideas or for documentation purposes. We currently have more than four technology variants, hence the need for more than four entries in the history list.

klayoutmatthias, 2019-11-08:
Thanks for this nice feedback :-)

However, XSection is a separate (and basically independent) project. Would you mind taking this request there?

XSection's GitHub repo is here: https://github.com/klayoutmatthias/xsection

Thanks,

Matthias

cfriesicke, 2019-11-10:
Yes, I will open the issue there, so it can be closed here.

closed 2019-11-10 by cfriesicke