KLayout/klayout #405: latest pip package fails to load submodules on Windows

simbilod, 2019-11-08:
Using Anaconda3 with Python 3.6.8, if I "pip install klayout" I cannot import, say, klayout.db.

If I revert to an earlier version e.g. "pip install klayout==0.26.0.dev15", everything works fine.

klayoutmatthias, 2019-11-11:
Frankly, this version should not be released yet. I have removed 0.26.1 and will replace that by a new version one it's confirmed that this one works.

thomaslima, 2019-11-11:
Matthias. The failing version is the 0.26.0, not 0.26.1. I think we should remove that as well? PyPI doesn't like removing released packages. Maybe we should upload to the test pypi unless we are ready to ship.

klayoutmatthias, 2019-11-11:
Thanks for clarifying this. 

But I can't should not remove all of 0.26.0 because some packages are actually good.

What exactly is failing with 0.26.0?

Matthias

klayoutmatthias, 2019-11-12:
I'm not entirely sure what's the problem - the 0.26.1 I pushed was broken because it wasn't the build from the Azure pipeline. Mine did not have the wheel patched with the helper DLLs. I tried to fix that with data_files, but that's not doing what I wanted (it places files relative to sys.prefix, but not relative to the package).

But 0.26.0 should be working as it includes the DLLs. Maybe the VS runtime is missing. Could you try to install it (e.g. from https://support.microsoft.com/en-us/help/2977003/the-latest-supported-visual-c-downloads) and check if it works with 0.26.0? 

Thanks,

Matthias

klayoutmatthias, 2020-02-21:
Can I close this issue?

simbilod, 2020-02-22:
Sorry, haven't taken the time to tinker with this. As disclosed before, the version I pip'ed above works.

klayoutmatthias, 2020-02-22:
It's Ok ... I wasn't expecting this. I'm just cleaning up my open tickets. I'll close it then as there are many new (and working) packages now.

closed 2020-02-22 by klayoutmatthias