KLayout/klayout #416: Table tab of User properties

EugeneZelenko, 2019-11-18:
I tried 0.26.1 build for Ububtu 18 from https://www.klayout.de/build.html, but it seems that +, - and "Edit" buttons are not working. "Text Editor" tab works fine.

Will be good idea to handle double click on table as "Edit" shortcut.

klayoutmatthias, 2019-11-19:
Hi Eugene,

I can't confirm this - I'm using Ubuntu 18 myself :(

Maybe it's dependent on the window manager - i.e. the property key/value dialog is appearing, but hidden somewhere?

I'm using Gnome.

Matthias

klayoutmatthias, 2019-11-19:
Hi Eugene,

I just tried with a Docker container based on Ubuntu 18.04 and the RPM from the download page and it works as expected :(

This is how I did this:

Start the docker container with:

```
xhost + && docker run -ti --rm -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v $HOME/.Xauthority:/home/developer/.Xauthority --net=host ubuntu:18.04
```

Inside the container install KLayout:

```
apt-get update
apt-get install wget
wget https://www.klayout.org/downloads/Ubuntu-18/klayout_0.26.1-1_amd64.deb
apt install ./klayout_0.26.1-1_amd64.deb
QT_X11_NO_MITSHM=1 klayout  -e
```

Matthias

EugeneZelenko, 2019-11-19:
Hi, Matthias!

Dialog appears as expected, but "+", "-" and "Edit" buttons on "Table" page doesn't produce expected effects.

klayoutmatthias, 2019-11-20:
Is there an error message on the terminal maybe?

I really don't know what to look for :(

Matthias

EugeneZelenko, 2019-11-20:
Looks my layout is in read-only mode. I think will be good idea to disable "+", "-" and "Edit" buttons when m_editable is false. At least this will reduce users' confusion :-)

Eugene.

closed 2019-12-07 by klayoutmatthias

EugeneZelenko, 2019-12-08:
Thank you for fix!