KLayout/klayout #425: Dummy pins generated for dummy gates

klayoutmatthias, 2019-11-21:
For details see here: https://www.klayout.de/forum/discussion/1375/extra-pins-are-extracted-at-dummy-poly-data#latest

klayoutmatthias, 2019-11-21:
[issue-425.zip](https://github.com/KLayout/klayout/files/3877068/issue-425.zip)

The attached layout demonstrates the issue with this netlist extraction script (run as LVS - does not product a cross-reference, but only a netlist):

```

# Enable hierarchical mode
deep

report_lvs

# -------------------------------------------------------------------
# Layers

# Drawing layers
nwell       = input(1, 0)
active      = input(2, 0)
poly        = input(3, 0)
poly_lbl    = input(3, 1)
diffcont    = input(4, 0)
polycont    = input(5, 0)
metal1      = input(6, 0)
metal1_lbl  = input(6, 1)
via1        = input(7, 0)
metal2      = input(8, 0)
metal2_lbl  = input(8, 1)

# Bulk layer for terminal provisioning
bulk        = polygon_layer

# Computed layers
pactive               = active & nwell
pgate                 = pactive & poly
psd                   = pactive - pgate

nactive               = active - nwell
ngate                 = nactive & poly
nsd                   = nactive - ngate

# -------------------------------------------------------------------
# Device extraction

# PMOS transistor device extraction
extract_devices(mos3("PMOS"), { "SD" => psd, "G" => pgate, 
                                "tS" => psd, "tD" => psd, "tG" => poly })

# NMOS transistor device extraction
extract_devices(mos3("NMOS"), { "SD" => nsd, "G" => ngate, 
                                "tS" => nsd, "tD" => nsd, "tG" => poly })

# -------------------------------------------------------------------
# Connectivity

# Inter-layer
connect(psd,        diffcont)
connect(nsd,        diffcont)
connect(poly,       polycont)
connect(poly,       poly_lbl)
connect(polycont,   metal1)
connect(diffcont,   metal1)
connect(metal1,     metal1_lbl)
connect(metal1,     via1)
connect(via1,       metal2)
connect(metal2,     metal2_lbl)

# -------------------------------------------------------------------
# Netlist and compare

# Netlist normalization
netlist.simplify
```

The effect is: the INV2X cell will have four dummy pins corresponding to the four dummy gates inside each INV2X.

closed 2019-12-18 by klayoutmatthias

jensen612, 2022-04-09:
Hi Matthias, I just came across your post here. I'm seeing a similar issue as dummy pins generated for shared dummy gates, when doing RC extraction. But I'm using the Calibre+StarRC flow, which is different with your flow. I don't quite understand your solutions to this issue, and I wonder if by any chance, you have the experience to help fix my issue? Thank you very much!