KLayout/klayout #460: MacOS build for 0.26.2

Kazzz-S, 2020-01-02:
Hi Matthias,

## A Happy New Year, 2020! ##
I could somehow catch up with the progress :-).
Now, different DMG packages of **KLayout 0.26.2 for Mac** are available as below ( [Dropbox Link](https://www.dropbox.com/sh/or68i4qhzjrz0v9/AAC4y8BxAZRrXnWsZHSNMdoPa?dl=0) ).

|Srl.No.|Target OS (1)|DMG file name (2) |Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
1|ElCapitan (10.11)|ST-klayout-0.26.2-MacOSX-ElCapitan-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled| (3) |
2|Sierra (10.12)|ST-klayout-0.26.2-macOS-Sierra-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(3)|
3|HighSierra ( 10.13)|ST-klayout-0.26.2-macOS-HighSierra-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(3)|
4|Mojave (10.14)|ST-klayout-0.26.2-macOS-Mojave-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(3)|
5|Catalina (10.15)|ST-klayout-0.26.2-macOS-Catalina-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(3)|

(1) Yosemite (10.10) has been dropped due to the compiler issue reported in #360.
(2) DMG file naming convention has been changed, as shown below.
[image]

where
(a) Type of package:  ST-: standard; LW-: lightweight; HW-: heavyweight
(b) KLayout version
(c) Target OS
(d) Package serial number
(e) Qt5 type:  qt5MP: MacPorts; qt5Brew: Homebrew; qt5Ana3: Anaconda3
(f) **R**uby type:    sys: OS-bundled; hb26: Ruby2.6 from Homebrew; ana3: Ruby2.5 from Anaconda3
(f) **P**ython type: sys: OS-bundled; hb37: Python 3.7 from Homebrew; ana3: Python3.7 from Anaconda3

(3) Only Qt's frameworks are included in the DMG package, which is the standard (ST-).

----
The three DMG packages below are experimental and available only for Catalina (10.15) but may be useful for some users.

|Srl.No.|Target OS|DMG file name (2) |Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
6|Catalina (10.15)|LW-klayout-0.26.2-macOS-Catalina-1-qt5Brew-Rhb26Phb37.dmg|Homebrew|Homebrew|Homebrew| (4) |
7|Catalina (10.15)|LW-klayout-0.26.2-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(5)|
8|Catalina (10.15)|HW-klayout-0.26.2-macOS-Catalina-1-qt5Brew-RsysPhb37.dmg|Homebrew|OS-bundled|Homebrew|(6)|

(4) None of Qt, Ruby, and Python frameworks are included in the DMG, hence, lightweight. This package is for those who have a Homebrew development environment.

(5) None of Qt, Ruby, and Python frameworks are included in the DMG, hence, lightweight. This package is for those who have an Anaconda3 development environment.
* I'm mainly using this package with a Bash script bundle in `script-bundles.zip`.

(6) Both Qt5 and Python3.7 frameworks are included. Hence, heavyweight, but the OS-bundled Ruby is not included in the package.

With warm regards,
Kazzz

Kazzz-S, 2020-01-06:
Started building 0.26.3 ( 80b0eae)...

klayoutmatthias, 2020-01-06:
Wow! Thanks a lot and a happy 2020 too!! :)

I'm sorry I had to release 0.26.3. I'll copy over the 0.26.2 versions and will do the same for the 0.26.3 package once it's available.

Many thanks for your continuous support!

Best regards,

Matthias

klayoutmatthias, 2020-01-06:
Thanks again! :)

BTW: I just published the 0.26.2 builds ... I hope I got the explanations right. If that's fine, I'll follow the same scheme for 0.26.3. 

And thanks for the excellent description above!

Best regards,

Matthias

klayoutmatthias, 2020-01-07:
Hi kazzz, 

sorry to bother you but there is one thing that should be fixed: the buddy tools can't find the streamer plugins. The reason is that these tools look up the db_plugins directory with the streams next to the libklayout_db.0.dylib files. 

I got a user error report for this. As a workaround, I was able to fix this by setting a relative symbolic link:

```
cd /Applications/klayout.app/Contents/Frameworks
ln -s ../MacOS/db_plugins db_plugins
```

will it be possible to include this into the DMG?

Thanks and best regards,

Matthias

Kazzz-S, 2020-01-07:
> Hi kazzz,
>
> sorry to bother you but there is one thing that should be fixed: the buddy tools can't find the streamer plugins. The reason is that these tools look up the db_plugins directory with the [...]

Hi Matthias, 
I think it should be possible. Let me try.
Please instruct how I should test it using which buddy tools.

Regards,
Kazzz

martinbxx, 2020-01-08:
I am the user who reported the issue (and who didn't know about this github site). To reproduce:
/Applications/klayout.app/Contents/Buddy/strm2oas swan_min.gdstxt swan_min.oas
I am going to attach the (small) swan_min.gdstxt file (edit: I had to rename to swan_min.txt because it wouldn't let me upload with extension gdstxt.)

Thanks for the Catalina compiles. And yes, I am using your LW-klayout-0.26.2-macOS-Catalina-1-qt5Brew-Rhb26Phb37.dmg.

[swan_autoss.txt](https://github.com/KLayout/klayout/files/4033020/swan_autoss.txt)

Martin

Kazzz-S, 2020-01-08:
Hi Martin,

Thank you for proving the test case, and sorry for the inconvenience caused.
With the test case, I could reproduce the issue.
I have also confirmed that the workaround suggested by Matthias works fine.
So I'll modify the DMG maker accordingly and remake DMGs.
Once it's ready, I'll inform you.

Regards,
Kazzz

Kazzz-S, 2020-01-08:
Hi Martin,

**LW-klayout-0.26.3-macOS-Catalina-1-qt5Brew-Rhb26Phb37.dmg** is ready [here](https://www.dropbox.com/sh/euttu8cl4hffahc/AADCSMjg7AvoL7Ooj52HPhX2a?dl=0).
Please try this one and let me know if this is OK with you.

Also, please note that this is a temporary one. 
After committing the changes, I'm going to rebuild the tools to incorporate the latest revision info.
Then I'll replace the DMG, too.

Regards,
Kazzz

martinbxx, 2020-01-08:
Hi Kazzz,

thanks so much for the quick response. I should have mentioned that I already verified the fix with the symlink that Matthias provided. I installed your new version and verified that it now works. Thanks very much! I was very unhappy last fall when I installed Catalina and realized that klayout was broken. It helps me a lot to have it work again.

Also, your packaging is very nice. It is very easy to use and install.

Martin

Kazzz-S, 2020-01-08:
> Thanks again! :)
> 
> BTW: I just published the 0.26.2 builds ... I hope I got the explanations right. If that's fine, I'll follow the same scheme for 0.26.3.
> 
> And thanks for the excellent description above!
> 
> Best regards,
> 
> Matthias

Hi Matthias,

Thanks for publishing 0.26.2.-DMGs.
I feel explaining the package features in a short note is NOT easy. That's why the long description above.

I would suggest:
 * For those five **ST-*.dmg**, drop "MacPorts based" because no development environment is needed.

For remaining three DMGs 
 * LW-klayout-0.26.2-macOS-Catalina-1-qt5Brew-Rhb26Phb37.dmg (Catalina (10.15) HomeBrew based - experimental)
* LW-klayout-0.26.2-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg (Catalina (10.15) Anaconda based - experimental) (as it is)
 * HW-klayout-0.26.2-macOS-Catalina-1-qt5Brew-RsysPhb37.dmg (Catalina (10.15)  HomeBrew Python3 included - experimental)

Once 0.26.3 DMGs are ready, I'll inform you.
Regards,
Kazzz

Kazzz-S, 2020-01-08:
> Hi Kazzz, thanks so much for the quick response. I should have mentioned that I already verified the fix with the symlink that Matthias provided. I installed your new version and verified that it [...]

Hi Martin,

Thanks for testing the new package. Nice to know that it is useful for you.
I'll proceed.

Regards,
Kazzz

klayoutmatthias, 2020-01-08:
Thanks a lot ... I updated the description texts according to your suggestions.

Many thanks for your efforts and best regards,

Matthias

Kazzz-S, 2020-01-09:
Hi Matthias,

Sorry to bother you. My idea above was a bit ambiguous.
I think the modified explanation shown below is better.
[image]

With warm regards,
Kazzz

klayoutmatthias, 2020-01-09:
No worries, iterations are OK :)

I changed it accordingly. Thanks for the clear description! The texts are generated and confined to single lines, so I could not include the footnote. But I added the text behind the download link:

[image]

Best regards,

Matthias

Kazzz-S, 2020-01-10:
> No worries, iterations are OK :)
> 
> I changed it accordingly. Thanks for the clear description! The texts are generated and confined to single lines, so I could not include the footnote. But I added the text behind the download link:
> 
> [image]
> 
> Best regards,
> 
> Matthias

Hi Matthias,

Thanks for modifying the short notes on DMG packages.
**0.26.3-DMGs** will be ready shortly.

Warm regards,
Kazzz

Kazzz-S, 2020-01-10:
## Mac DMGs for 0.26.3 ##

Hi Matthias,

Different DMG packages of **KLayout 0.26.3 for Mac** are available as below ( [0.26.3 Dropbox Link](https://www.dropbox.com/sh/euttu8cl4hffahc/AADCSMjg7AvoL7Ooj52HPhX2a?dl=0) ).
**Please note that _pymod_ is not included in those packages.**

|Srl.No.|Target OS (1)|DMG file name (2) |Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
1|ElCapitan (10.11)|ST-klayout-0.26.3-MacOSX-ElCapitan-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled| (3) |
2|Sierra (10.12)|ST-klayout-0.26.3-macOS-Sierra-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(3)|
3|HighSierra ( 10.13)|ST-klayout-0.26.3-macOS-HighSierra-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(3)|
4|Mojave (10.14)|ST-klayout-0.26.3-macOS-Mojave-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(3)|
5|Catalina (10.15)|ST-klayout-0.26.3-macOS-Catalina-1-qt5MP-RsysPsys.dmg|MacPorts|OS-bundled|OS-bundled|(3)|

(1) Yosemite (10.10) has been dropped due to the compiler issue reported in #360.
(2) DMG file naming convention has been changed from earlier (< 0.26.x) releases, as shown below.
**0.26.2** is taken as an example here.
[image]

where
(a) Type of package:  ST-: standard; LW-: lightweight; HW-: heavyweight
(b) KLayout version
(c) Target OS
(d) Package serial number
(e) Qt5 type:  qt5MP: MacPorts; qt5Brew: Homebrew; qt5Ana3: Anaconda3
(f) **R**uby type:    sys: OS-bundled; hb26: Ruby2.6 from Homebrew; ana3: Ruby2.5 from Anaconda3
(f) **P**ython type: sys: OS-bundled; hb37: Python 3.7 from Homebrew; ana3: Python3.7 from Anaconda3

(3) Only Qt's frameworks are included in the DMG package, which is the standard (ST-).

----
The three DMG packages below are experimental and available only for Catalina (10.15) but may be useful for some users.

|Srl.No.|Target OS|DMG file name (2) |Qt5|Ruby|Python|Remarks|
|:------:|:--------:|:--------------:|:---:|:----:|:-----:|:--------:|
6|Catalina (10.15)|HW-klayout-0.26.3-macOS-Catalina-1-qt5Brew-RsysPhb37.dmg|Homebrew|OS-bundled|Homebrew|(4)|
7|Catalina (10.15) with Homebrew dev. env.|LW-klayout-0.26.3-macOS-Catalina-1-qt5Brew-Rhb26Phb37.dmg|Homebrew|Homebrew|Homebrew| (5) |
8|Catalina (10.15) with Anaconda3 dev. env.|LW-klayout-0.26.3-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg|Anaconda3|Anaconda3|Anaconda3|(6)|

(4) Both Qt5 and Python3.7 frameworks are included, hence, heavyweight. As for Ruby, the OS-bundled one is used.

(5) None of Qt, Ruby, and Python frameworks are included in the DMG, hence, lightweight. This package is for those who have a Homebrew development environment.

(6) None of Qt, Ruby, and Python frameworks are included in the DMG, hence, lightweight. This package is for those who have an Anaconda3 development environment.
* I'm mainly using this package with a Bash script bundle in `script-bundles.zip`.

With warm regards,
Kazzz

P.S. Please close this thicket after publishing the packages.

klayoutmatthias, 2020-01-10:
Files are on the server. Many thanks for these efforts!

lukasc-ubc, 2020-02-19:
Hi @Kazzz-S 

I just upgraded to Catalina, and trying the various flavours:

1) Regular Catalina version, which comes with Python2. Works, but isn't Python 3...

2) Homebrew Python 3 version: pip install doesn't work. missing openSSL

3) Anaconda3 version: I installed Anaconda 3.7 (latest version).  I get an error as follows: 
> dyld: Library not loaded: @rpath/libpython3.7m.dylib

Any thoughts?

thank you
Lukas

Kazzz-S, 2020-02-20:
Hi @lukasc-ubc ,

Thanks for your response.

## 1. Regular Catalina version: ##
**ST-klayout-0.26.3-macOS-Catalina-1-qt5MP-RsysPsys.dmg**
Yes, Python is still 2.7, which is the OS-bundled one.
By purchasing a new Mac :-), I have jumped from Yosemite to Catalina expecting that
Python 3.x is bundled. However, ... :-( Nevertheless, I'm using Yosemite, too.
Irrespective of our strong expectations, the **standard** is still Python 2.7
on different macOS. So, in my opinion, this KLayout version is also the **standard**.
But, apparently, this version is not convenient for some power users, and different users have
different preferences.

## 2. Homebrew Python 3 version: ##
**HW-klayout-0.26.3-macOS-Catalina-1-qt5Brew-RsysPhb37.dmg**
To be honest, I'm not a Homebrew user. I have manually merged changes made earlier in the
package building tools for Mac and could successfully build this package.
However, I have not tested **"pip"** with Homebrew. Therefore, truly **experimental**.
I want to request help from the community regarding this.

My Homebrew environment is minimal to build KLayout, as shown below.

```
(base) MacBookPro2{kazzz-S}(1)$ pwd
/usr/local/Cellar

(base) MacBookPro2{kazzz-S}(2)$ ll
Total 0
drwxrwxr-x 13 kazzz-S admin 416  1 25 10:16 .
drwxr-xr-x 16 root     wheel 512  1 29 18:00 ..
-rw-r--r--  1 kazzz-S admin   0 11 24 17:22 .keepme
drwxr-xr-x  3 kazzz-S staff  96 11 24 17:36 gdbm
drwxr-xr-x  3 kazzz-S staff  96 11 24 17:38 libyaml
drwxr-xr-x  3 kazzz-S staff  96  2  7 06:58 mono
drwxr-xr-x  3 kazzz-S staff  96 11 24 17:36 openssl@1.1
drwxr-xr-x  3 kazzz-S staff  96  2  5 06:36 python
drwxr-xr-x  3 kazzz-S staff  96  1 29 19:10 qt
drwxr-xr-x  3 kazzz-S staff  96 11 24 17:36 readline
drwxr-xr-x  3 kazzz-S staff  96  2 18 06:42 ruby
drwxr-xr-x  3 kazzz-S staff  96  2  5 06:36 sqlite
drwxr-xr-x  3 kazzz-S staff  96 11 24 17:36 xz
```

```
(base) MacBookPro2{Kazzz-S}(3)$ pwd
/usr/local/opt
(base) MacBookPro2{Kazzz-S}(4)$ ll
Total 0
drwxrwxr-x 17 Kazzz-S admin 544  2 18 06:42 .
drwxr-xr-x 16 root     wheel 512  1 29 18:00 ..
-rw-r--r--  1 Kazzz-S admin   0 11 24 17:22 .keepme
lrwxr-xr-x  1 Kazzz-S admin  21 11 24 17:36 gdbm -> ../Cellar/gdbm/1.18.1
lrwxr-xr-x  1 Kazzz-S admin  23 11 24 17:38 libyaml -> ../Cellar/libyaml/0.2.2
lrwxr-xr-x  1 Kazzz-S admin  24  2  7 06:58 mono -> ../Cellar/mono/6.8.0.105
lrwxr-xr-x  1 Kazzz-S admin  28 11 24 17:36 openssl -> ../Cellar/openssl@1.1/1.1.1d
lrwxr-xr-x  1 Kazzz-S admin  28 11 24 17:36 openssl@1.1 -> ../Cellar/openssl@1.1/1.1.1d
lrwxr-xr-x  1 Kazzz-S admin  24  2  5 06:36 python -> ../Cellar/python/3.7.6_1
lrwxr-xr-x  1 Kazzz-S admin  24  2  5 06:36 python3 -> ../Cellar/python/3.7.6_1
lrwxr-xr-x  1 Kazzz-S admin  19  1 29 19:10 qt -> ../Cellar/qt/5.14.1
lrwxr-xr-x  1 Kazzz-S admin  19  1 29 19:10 qt5 -> ../Cellar/qt/5.14.1
lrwxr-xr-x  1 Kazzz-S admin  24 11 24 17:36 readline -> ../Cellar/readline/8.0.1
lrwxr-xr-x  1 Kazzz-S admin  20  2 18 06:42 ruby -> ../Cellar/ruby/2.7.0
lrwxr-xr-x  1 Kazzz-S admin  23  2  5 06:36 sqlite -> ../Cellar/sqlite/3.31.1
lrwxr-xr-x  1 Kazzz-S admin  23  2  5 06:36 sqlite3 -> ../Cellar/sqlite/3.31.1
lrwxr-xr-x  1 Kazzz-S admin  18 11 24 17:36 xz -> ../Cellar/xz/5.2.4
```

By the way, what would happen with **LW-klayout-0.26.3-macOS-Catalina-1-qt5Brew-Rhb26Phb37.dmg**?
This version also may fail because Homebrew very recently upgraded both Python and Ruby.
More precicely,

```
Jan.2020 ---> as of Feb. 18th 2020
Ruby 2.6.5 ---> 2.7.0
Python 3.7.5 ---> 3.7.6
```

This package was built with the previous Python and Ruby.

## 3. Anaconda3 version: ##
**LW-klayout-0.26.3-macOS-Catalina-1-qt5Ana3-Rana3Pana3.dmg**
This version is my primary environment at home, which works fine so far.
I've used **Anaconda 2019.10 for macOS Installer**.
I believe your environment is the same as mine. So I'm a bit puzzled.
One thing I noticed while the installation is that earlier, the default installation location was
**$HOME/anaconda3** but it is now **$HOME/opt/anaconda3**

I can manage Python packages with **conda** command comfortably.
From KLayout PYA, I often use Pandas to dump some internal data structure to CSV files,
which works pretty fine.
To use this version, I have made a simple script bundle using **Automator.app** tool that
invokes a simple Bash script as below:

```
export PYTHONHOME=$HOME/opt/anaconda3
myconfig=$HOME/.klayout/klayoutrc-ana3
open -n -a /Applications/klayout-ana3.app --args -e -c $myconfig -style=fusion
```

Please note I have renamed "klayout.app" to "klayout-ana3.app" under /Applications.

Thank you
Kazzz

Kazzz-S, 2020-03-07:
Close this ticket since version [0.26.4](https://www.klayout.de/development.html#0.26.4) has been released.
Please also refer to #519.

closed 2020-03-07 by Kazzz-S