KLayout/klayout #472: pypi version not building/up to date?

beuerle, 2020-01-08:
As of now, I am not able to install `klayout` via pip on Linux:

```bash
$ pip install --user klayout
Collecting klayout
  Using cached https://files.pythonhosted.org/packages/3e/e2/9415684a50f9a512f6139cd55b3f8c9ba460b5c190400b7bd764de64ce04/klayout-0.26.0.dev11.tar.gz
Building wheels for collected packages: klayout
[...]
```

The complete log output can be found [here](https://gist.github.com/beuerle/5370c5a19a8f41a43f4b4d636624d0a3).

First of all, it tries to download `klayout-0.26.0.dev11` instead of the most current version.
According to the [release history](https://pypi.org/project/klayout/) on pypi `v0.26.3` should be available:

[image]

Searching also gives `0.26.3` as the current revision:

```bash
$ pip search klayout                           
klayout (0.26.3)    - KLayout standalone Python package
```

However, trying to explicitly install `0.26.3` fails:

```bash
$ pip install --user klayout==0.26.3
Collecting klayout==0.26.3
  ERROR: Could not find a version that satisfies the requirement klayout==0.26.3
(from versions: 0.26.0.dev2, 0.26.0.dev3, 0.26.0.dev4, 0.26.0.dev9, 0.26.0.dev10, 0.26.0.dev11)
ERROR: No matching distribution found for klayout==0.26.3
```

Any ideas what's happening here?

klayoutmatthias, 2020-01-08:
Looks like your system isn't on the list. What's the distribution you're using?

I tested with Ubuntu 18.04 in a docker container:

Python 3:

```
$ docker run -it --rm ubuntu:18.04 bash
root@a7c7d5bdd1ef:/# apt-get update
...
root@a7c7d5bdd1ef:/# apt-get install python3-pip
...
root@a7c7d5bdd1ef:/# pip3 install klayout
Collecting klayout
  Downloading https://files.pythonhosted.org/packages/e5/c0/6e8893fb071cc87ec122e64ad4b307ed1ef0a78b46e7c95859716b33d03f/klayout-0.26.3-cp36-cp36m-manylinux1_x86_64.whl (11.2MB)
    100% |################################| 11.2MB 148kB/s 
Installing collected packages: klayout
Successfully installed klayout-0.26.3
root@a7c7d5bdd1ef:/# python3 
Python 3.6.9 (default, Nov  7 2019, 10:44:02) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import klayout.db
>>> print(str(klayout.db.Point(1, 2)))
1,2
```

Python 2:

```
$ docker run -it --rm ubuntu:18.04 bash
root@1f99cc572682:/# apt-get update
...
root@1f99cc572682:/# apt-get install python-pip
...
root@1f99cc572682:/# pip install klayout       
Collecting klayout
  Downloading https://files.pythonhosted.org/packages/1a/4d/ed6bbf7a1994ed97c7fbde94cbe02849d6cc034d555f41c523ab22ab45f6/klayout-0.26.3-cp27-cp27mu-manylinux1_x86_64.whl (11.2MB)
    100% |################################| 11.2MB 129kB/s 
Installing collected packages: klayout
Successfully installed klayout-0.26.3
root@1f99cc572682:/# python 
Python 2.7.17 (default, Nov  7 2019, 10:07:09) 
[GCC 7.4.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import klayout.db
>>> print(str(klayout.db.Point(1, 2)))
1,2
```

So both ways work.

beuerle, 2020-01-08:
I'm running Arch Linux, and the problem is that I already have python 3.8 installed:

```bash
$ python --version
Python 3.8.1
```

So I probably have to run my layout scripts in a virtual environment until klayout builds with python 3.8.

thomaslima, 2020-10-23:
A not so elegant solution to this is to build from source. @beuerle if you clone this repo or download the zip file for the latest release, you can navigate to the cloned folder and run `pip install .`.

@klayoutmatthias pypi will attempt to install from source if the source distribution is present. Version 0.26.3's source seems to be missing both the source and the wheels for 3.8 but they are there in the most recent versions.

I recommend closing this.

closed 2021-08-24 by klayoutmatthias