KLayout/klayout #475: possible feature: allignment for CAM data

zanglerecke, 2020-01-10:
Last Year (just some days ago) i had to compare some old CAM output from the manufacturer, and the one of the adapted design. 

The Task was clear: just align both correct layers and compare…

the compare of the output of the old layout and the new one was easy… they came aligned / I exported them with the same settings.

But from the Manufacturer we get the whole Frame.

So the Idea: just mark everything on a specific Layer, and then select two identical objects and snap them together. → everything else should follow the same Distance.

Maybe i just did not see this function, but if it is missing, it could have made my life much more easy.

By the way…. Thanks for your beautiful AWESOME tool.

klayoutmatthias, 2020-01-10:
Thanks for this nice feedback :)

For the feature request: you mean those objects are on two different layouts? Or different layers of the same layout?

Matthias

zanglerecke, 2020-01-11:
"both"... i have separate files for every layer of my PCB....i import them as separate layers. 
one layer = one Gerber-File. 
as i have e.g. 2 CAM-exports=2PCBs with e.g. 8 layers each. 

so i have totally 16 layers but 2 designs with 8 layers each. 
normally all 8 layers "should" be aligned, but sometimes they are not. (very old Data)

on nearly every layout are fiducials .. circles to align the ?automatic assembly line? (German: Bestückungsautomaten)

and sometimes those layers are mirrored ... (e.g. for Backside, or Masking layers)

demisjohn, 2020-05-24:
Seems like you could write a script to do this, which then gives you flexibility of which layers/layouts you want to be transformed by the Move command.  I'm not sure how you tell a script which particular objects to align though, but iterating through a layout's Layers and moving all objects sounds pretty straightforward.