KLayout/klayout #479: import klayout.lay

tkoyat, 2020-01-19:
It seems that there is no way to import the lay package

klayoutmatthias, 2020-01-22:
You mean from PyPI?

The klayout package from PyPI only supports tl, db, lib, rdb. lay requires Qt as it provides the user interface and that's a too big overhead for PyPI. Plus there are license implications I want to avoid. 

The Linux distributions of klayout come with a full klayout Python package and link against Qt installations from the Linux system. Those modules are not distributed through PyPI however.

Matthias

tkoyat, 2020-02-01:
Hi Matthias :
 i build klayout-0.26.3 success, want to import pya for show layout, but have the following errors

>>> import pya
>>> app = pya.Application.instance()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'pya' has no attribute 'Application'
>>>

tkoyat, 2020-02-01:
Hi ,Matthias:
 I need to use python to show mainwindow , can you help me provide some reference code?

klayoutmatthias, 2020-02-03:
Sorry, this won't work.

There are basically three modes:

1. bare metal - this is the Python module you find on PyPI. It only supplies the non-UI-related classes (db, lay, lib, tl, rdb). It works without Qt.
2. Python module with UI classes: this is what you get with the Linux build or installation. It supplies the lay module and the Qt binding classes. You can basically build UI applications, but the applications are somewhat limited. As in this mode the application is the Python interpreter there is no KLayout Application. There also isn't a MainWindow. You cannot instantiate one currently.
3. KLayout as the Python interpreter. For this mode, run KLayout with "klayout -z -r yourpythonscript.py". The Python module is "pya". Only this mode will supply pya.Application and pya.MainWindow.

So bottom line is: the Python modules will not give you MainWindow. Use KLayout with "-z" as the Python interpreter.

closed 2020-05-23 by klayoutmatthias

lukasc-ubc, 2020-05-23:
This is wonderful, Matthias. 

could we use the same “import” definitions in all three approaches you list?  

Namely, for #3, instead of 
 > import pya

could we have
 > import klayout

That way our scripts/PCells will work inside and out.

Otherwise, presumably we need something like 
try: import pya

Or some “if” statement 

thanks,
Lukas